noble/bleno #283: uuids is being cached

anchetaWern, 2017-03-19:
I'm not sure why but it seems like the uuids are being cached somewhere. Not sure if its the device or bleno itself but changing the uuids doesn't change what's actually being advertised (even though when I `console.log` from the Node side of things it's reflecting the new values). But when testing via a Cordova app, the old uuid is still being advertised. I'm also changing the advertisement name. Part of the name is used for constructing the uuid but the odd thing is that the advertisement name is changing while the uuid remains constant. This is why the app fails to connect because the actual uuid being advertised is no longer the same with the uuid being constructed by the app (as based from the advertisement name).

Here's what the code looks like:

```
var bleno = require('bleno');
var BlenoPrimaryService = bleno.PrimaryService;
bleno.on('stateChange', function(state) {
 
    if (state === 'poweredOn') {
        console.log('peripheral id: ' + config.peripheral_id);
        bleno.startAdvertising(config.name, [config.peripheral_id]);
    }
});

bleno.on('advertisingStart', function(error) {
    console.log('primary service uuid: ' + config.primaryservice_uuid); // shows the correct uuid but still advertising the old one
    if (!error) {
        bleno.setServices([
            new BlenoPrimaryService({
            uuid: config.primaryservice_uuid,
                characteristics: [some characteristics here]
           })
        ]);
    }
});
```

I already tried the following but nothing seems to work:

- restarting the Pi (`sudo shutdown -r now`)
- shutting down the Pi then turning on again (`sudo shutdown -h now`)
- shutting down the Pi and then unplugging, then plug again
- clear the cache of the Cordova app.
- uninstalling the Cordova app and reinstalling again.

Has anyone ever encountered this issue? Anything else that I could try or somewhere I've missed? Thanks in advance.

storm1er, 2017-03-20:
Got same issue, I think this is why I can't write to my RPi https://github.com/sandeepmistry/bleno/issues/280

sandeepmistry, 2017-03-27:
@anchetaWern could you provide an HCI dump capture for this? `sudo hcidump -t -x`

What version of Android are you using? Could you also try turning bluetooth off and on on the Android device?

jmrgibson, 2017-04-05:
I had the same problem, running bleno on mac with an android device. Turns out Android is the one doing the cashing, starting the bluetooth on the android device fixed the problem

brycejacobs, 2017-04-25:
This is on purpose, and should happen. In most cases in today's BLE stacks, once you do a scan, it will cache the data unless a service changed packet comes across invalidating the service.

This is one of the reasons I wrote #128

closed 2018-01-29 by sandeepmistry