noble/bleno #284: Any luck with BeagleBone devices?

malikolivier, 2017-03-31:
I tried bleno on a BeagleBone Green Wireless running Debian.

I am using the factory version of the image with node 6.X.X:
```
BeagleBoard.org Debian Image 2016-06-14
```

Then I simply did:
```sh
# Install needed packages
$ apt-get install bluetooth bluez libbluetooth-dev libudev-dev
# Setup bluetooth device
$ bb-wl18xx-bluetooth
Found a Texas Instruments' chip!
Firmware file : /lib/firmware/ti-connectivity/TIInit_11.8.32.bts
Loaded BTS script version 1
texas: changing baud rate to 3000000, flow control to 1
Device setup complete
$ npm install bleno
# Enter node's command prompt:
$ node
> var bleno = require('bleno');
undefined
> bleno.state
'unknown'
```

No matter how much I wait, bleno state is always unknown, and advertising fails.
I tried run it with hciconfig. No luck either:

```
$ hciconfig hci0 leadv 0
Can't set advertise mode on hci0: Connection timed out (110)
```

Any one has had fun with BeagleBones before?

sandeepmistry, 2017-04-10:
@malikolivier I don't have a BeagleBone Green Wireless to test with.

Can you run an HCI dump capture `sudo hcidump -t -x` and share the output?

malikolivier, 2017-04-10:
Good news: It works now.

And I have no idea how. I just powered on my BBGW and tried to run BLE to send you the output, and it worked.

sandeepmistry, 2017-04-10:
Awesome, closing this as resolved then.

closed 2017-04-10 by sandeepmistry