noble/bleno #285: How to pass data bigger than 20bytes via notify/updateValueCallback ?

leviathan88, 2017-03-31:
First of all Bleno is great!,

My client (I am using nRF Connect app for testing it) can receive only 20bytes per message, 
for example if my this.message is 
`this.message="you are my sunshine my only sunshine"`
in read request here is how I send it back
```
onReadRequest(offset, callback) {
        if (offset) {
            callback(this.RESULT_SUCCESS, this.message.slice(offset))
        } else {              
            callback(this.RESULT_SUCCESS, this.message)
        }
    }
```

using it like this I manage to send the complete message,
but how can I achieve this in notify, that is with this._updateValueCallback ?

just to test it out, I tried a for loop, and sending chunks of data
`this._updateValueCallback(this.message.slice(startIndex,endIndex)`
but in the app (nRF Connect) I only receive/see the last part of the message,
in my case _" only sunshine"_,

how can I send the complete message ?

sandeepmistry, 2017-04-10:
@leviathan88 unfortunately notifications are limited by the MTU of the central.

You'll have to develop your own protocol to split up the data over notifications.

closed 2017-04-10 by sandeepmistry

leviathan88, 2017-04-12:
yeah figured that out :), either way thanks a lot !

CBZakaria, 2018-06-27:
Hello @leviathan88 did you solve your problem?

leviathan88, 2018-07-20:
Hi @CBZakaria  yes I did but don't remember how :/ sorry