noble/bleno #286: Tyring to use bleno within some promise library

bmwaechter, 2017-04-01:
Im trying to use bleno advertisements from within a promise resolution.  I.E. : 
```javascript
request(url)
  .then( function (body) {
     console.log(JSON.parse(body))
     config.UUID = JSON.parse(body).uuid
     console.log('---> New UUID: ' + config.UUID)
     bleno.on('stateChange', function(state) {
        console.log('on -> stateChange: ' + state)
        if (state === 'poweredOn'){
           bleno.startAdvertising(config.NAME, [config.UUID])
        } else {
           bleno.stopAdvertising()
        }
    })
  })
  .catch(function(err) {
     console.log(err)
     console.log('---> Keeping old UUID')
  })
```

I have tried countless iteration of this code, (ie. just calling advertisement start and other things like that).  But no matter what it seems to just make the request, then bleno just doesn't start.  period.  i must me misunderstanding something.  I would love any help, if not is this a bug? shouldn't we be able to start bleno/assign event handlers from in something like this?

sandeepmistry, 2017-04-10:
@bmwaechter you're probably missing the first state change event.

Could you check `bleno.state` in the request callback?

I can look into modifying bleno to not trigger the event until an `bleno.on('stateChange', ...)` is called ...

sandeepmistry, 2017-04-23:
Fixed via https://github.com/sandeepmistry/bleno/issues/293.

closed 2017-04-23 by sandeepmistry