noble/bleno #288: MTU max working feedbacks

farfromrefug, 2017-04-03:
Hi,

The main issue i am facing using bleno on multiple platforms is understanding what max MTU i can use.
For example on a raspberry Pi with any dongle, i can't get over 20. If i do the ble will die at one point and i have to restart the Pi.
Now this week end i just faced the same problem with the Edison (using Jubilinux and Yocto). 

First it would be nice to understand 2 things:
 * ```onSubscribe``` reports a max value of 101
 * ```mtu``` event reports a value of 104
Which one can i use as mtu size (packet size).

Now i tried to use 101, but the bluetooth was very unstable. I mean the first connexion worked great and i could send and receive (and i stressed it!) packets of size 101. But after one or 2 connect/disconnect, the bluetooth died and i had to restart the Edison.

The wish behind this issue is first to understand what ```onSubscribe`` and ```mtu``` values mean, and to find , through people experience the max actual MTU value we can use on each platform.

Maybe this could be put in the wiki?

sandeepmistry, 2017-04-10:
`onSubscribe` max value is `MTU` - 3 for the additional headers required for the notifications.

I suggest you looks at the HCI dump capture to better understand why the connection drops. It could be Bluetooth adapter specific too. Something like PR #268 might improve stability as well.

farfromrefug, 2017-04-11:
@sandeepmistry I got an hcidump with the issue. 
The process is this:
 - connect 
 - send a few packets
 - disconnect
 - connect
 - send a few packets 
=> the bluetooth then goes down on the Edison.
I don't see anything in the hcidump giving me clues on what's going on.
[Here](https://pastebin.com/gJEeE2wV) it is

Do you see anything obvious?

sandeepmistry, 2017-04-16:
Nothing obvious. Did you try stock bleno or with PR #268?

farfromrefug, 2017-04-17:
@sandeepmistry i tried stock. I am ok testing the PR but i need to understand. Do i still need to set the mtu? Do i need to limit the size of the packets i write?
Right now i splice the packets to the mtu size and send them in queue.

I want to test the PR the right way!
Thanks

sandeepmistry, 2017-04-17:
You should never set the MTU manually, it's something that is negotiated for you.

The notification data should never exceed the max size parameter of the `onSubscribe` callback.

farfromrefug, 2017-04-17:
@sandeepmistry Sorry if i explained my self uncorrectly. I don't ever change bleno's mtu variable ;)

Anyway just tried with the PR: 
* on a Edison with ```onSubscribe``` reporting a maxMtu of 101, and sending packages of size 101 (waiting for ```indicate``` to send the next one)

Everything seems to work PERFECTLY!!!! I have gone through 10 consecutive connections with big data sending, without any issue (ble was dropping around the second or third connection).

So for now that PR is the solution.
Will test more with other platforms and report here

Thank you so much!

sandeepmistry, 2017-04-24:
@farfromrefug thanks for trying it out!

I've merged https://github.com/sandeepmistry/bleno/pull/268, so closing this!

closed 2017-04-24 by sandeepmistry