noble/bleno #290: Gatt server add timeout to connected phone?

tnmendes, 2017-04-05:
I have used as base to develop the example of the BLE Pizza Service. When phone is connected to the GATT Server I want to disconnect him after some seconds i can achieve that goal?

Thanks in advance

sandeepmistry, 2017-04-10:
@tnmendes just call this after a `setTimeout`: https://github.com/sandeepmistry/bleno#disconnect-client

That should do the trick.

closed 2017-04-10 by sandeepmistry

tnmendes, 2017-04-10:
Thank you very much @sandeepmistry 

I gonna leave here my solution to help other persons that may have the same problem.

```
var peripheralConnected = false;

bleno.on('accept', function(clientAddress){

        logger.info('Peripheral :: Connected :: '+clientAddress);
        peripheralConnected = true;
        setTimeout(function(){
          
                forceDisconnect();
        }, 5000);
});

bleno.on('disconnect', function(clientAddress){

        logger.info('Peripheral :: Disconnect :: just informing that was disconnect '+clientAddress);
        peripheralConnected = false;
});

function forceDisconnect(){

   logger.info('Peripheral :: forceDisconnect');
   if(peripheralConnected){

      bleno.disconnect();
      peripheralConnected = false;
   }
}
```