noble/bleno #294: cannot find Services when I run the example "echo", but if I run two "echo" at the same time, it works

YangXinlei, 2017-04-19:
I use LightBlue on my iPhone and run "echo" on my raspberry pi 3. When only one "echo" at running, LightBlue can't connect to it. Which always pop up an alert and tells me: "Timeout interrogation the peripheral".
But if I keep the "echo" at running and then open another Terminal and run another "echo", then my LightBlue can connect to it successfully. when I write a value to it, both of the Terminals' output looks like this:

```
$ sudo node main.js 
bleno - echo
on -> stateChange: poweredOn
on -> advertisingStart: success
EchoCharacteristic - onWriteRequest: value = 08
```

sandeepmistry, 2017-04-23:
@YangXinlei Have you stopped `bluetoothd` as per the read me? https://github.com/sandeepmistry/noble/pull/514

YangXinlei, 2017-06-09:
@sandeepmistry Sorry! I opened this issue and I forgot... I Just tried those commends to stop `bluetoothd`, But looks like the problem remains the same.  I don't know if this is a common problem or it's just me. Maybe someday I'll reinstall a new `Raspbian`  to see if the problem is still there.

sandeepmistry, 2017-06-18:
@YangXinlei did you verified `bluetoothd` was stopped, via something like `ps -eaf | grep bluetoothd` ?

sandeepmistry, 2018-01-29:
Closing do you lack of activity.

closed 2018-01-29 by sandeepmistry