noble/bleno #295: Raspberry Pi fail

lemonhall, 2017-04-23:
pi@raspberrypi:~/bleno$ npm install bleno
/home/pi/bleno
└── bleno@0.4.2

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@~0.1.4 (node_modules/bleno/node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/bleno/package.json'
npm WARN bleno No description
npm WARN bleno No repository field.
npm WARN bleno No README data
npm WARN bleno No license field.
pi@raspberrypi:~/bleno$ node index.js
bleno warning: adapter state unauthorized, please run as root or with sudo
               or see README for information on running without root/sudo:
               https://github.com/sandeepmistry/bleno#running-on-linux
^Cpi@raspberrypi:~/bleno$ sudo node index.js

module.js:356
  Module._extensions[extension](this, filename);
                               ^
Error: /home/pi/bleno/node_modules/bluetooth-hci-socket/build/Release/binding.node: undefined symbol: node_module_register
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/home/pi/bleno/node_modules/bluetooth-hci-socket/lib/native.js:3:15)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
pi@raspberrypi:~/bleno$
----------------------------------------------------------------------

with npm@4.2.0 /home/pi/.nvm/versions/node/v7.9.0/lib/node_modules/npm

which install by nvm

lemonhall, 2017-04-23:
```
pi@raspberrypi:~/bleno$ rm -rf node_modules/
pi@raspberrypi:~/bleno$ npm install bleno

> usb@1.2.0 install /home/pi/bleno/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download: https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v51-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for usb@1.2.0 and node@7.9.0 (node-v51 ABI) (falling back to source compile with node-gyp)
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
make: Entering directory '/home/pi/bleno/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
  AR(target) Release/obj.target/usb.a
  COPY Release/usb.a
  CXX(target) Release/obj.target/usb_bindings/src/node_usb.o
In file included from ../src/node_usb.h:21:0,
                 from ../src/node_usb.cc:1:
../src/helpers.h: In function ‘void setConst(v8::Local<v8::Object>, const char*, v8::Local<v8::Value>)’:
../src/helpers.h:18:74: warning: ‘bool v8::Object::ForceSet(v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’ is deprecated (declared at /home/pi/.node-gyp/7.9.0/include/node/v8.h:2955): Use CreateDataProperty / DefineOwnProperty [-Wdeprecated-declarations]
  obj->ForceSet(Nan::New<String>(name).ToLocalChecked(), value, CONST_PROP);
                                                                          ^
  CXX(target) Release/obj.target/usb_bindings/src/device.o
In file included from ../src/node_usb.h:21:0,
                 from ../src/device.cc:1:
../src/helpers.h: In function ‘void setConst(v8::Local<v8::Object>, const char*, v8::Local<v8::Value>)’:
../src/helpers.h:18:74: warning: ‘bool v8::Object::ForceSet(v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’ is deprecated (declared at /home/pi/.node-gyp/7.9.0/include/node/v8.h:2955): Use CreateDataProperty / DefineOwnProperty [-Wdeprecated-declarations]
  obj->ForceSet(Nan::New<String>(name).ToLocalChecked(), value, CONST_PROP);
                                                                          ^
../src/device.cc: In static member function ‘static v8::Local<v8::Object> Device::get(libusb_device*)’:
../src/device.cc:42:76: warning: ‘v8::Local<v8::Object> v8::Function::NewInstance(int, v8::Local<v8::Value>*) const’ is deprecated (declared at /home/pi/.node-gyp/7.9.0/include/node/v8.h:3657): Use maybe version [-Wdeprecated-declarations]
   Local<Object> obj = constructorHandle->GetFunction()->NewInstance(1, argv);
                                                                            ^
../src/device.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE deviceConstructor(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/device.cc:58:79: warning: ‘bool v8::Object::ForceSet(v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’ is deprecated (declared at /home/pi/.node-gyp/7.9.0/include/node/v8.h:2955): Use CreateDataProperty / DefineOwnProperty [-Wdeprecated-declarations]
   Nan::New<Uint32>((uint32_t) libusb_get_bus_number(self->device)), CONST_PROP);
                                                                               ^
../src/device.cc:60:83: warning: ‘bool v8::Object::ForceSet(v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’ is deprecated (declared at /home/pi/.node-gyp/7.9.0/include/node/v8.h:2955): Use CreateDataProperty / DefineOwnProperty [-Wdeprecated-declarations]
   Nan::New<Uint32>((uint32_t) libusb_get_device_address(self->device)), CONST_PROP);
                                                                                   ^
../src/device.cc:63:67: warning: ‘bool v8::Object::ForceSet(v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’ is deprecated (declared at /home/pi/.node-gyp/7.9.0/include/node/v8.h:2955): Use CreateDataProperty / DefineOwnProperty [-Wdeprecated-declarations]
  info.This()->ForceSet(V8SYM("deviceDescriptor"), v8dd, CONST_PROP);
                                                                   ^
../src/device.cc:5:85: warning: ‘bool v8::Object::ForceSet(v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’ is deprecated (declared at /home/pi/.node-gyp/7.9.0/include/node/v8.h:2955): Use CreateDataProperty / DefineOwnProperty [-Wdeprecated-declarations]
   TARGET->ForceSet(V8STR(#NAME), Nan::New<Uint32>((uint32_t) (STR).NAME), CONST_PROP);
                                                                                     ^
../src/device.cc:68:2: note: in expansion of macro ‘STRUCT_TO_V8’
  STRUCT_TO_V8(v8dd, dd, bLength)
  ^
[... cut ...]
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@~0.1.4 (node_modules/bleno/node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/bleno/package.json'
npm WARN bleno No description
npm WARN bleno No repository field.
npm WARN bleno No README data
npm WARN bleno No license field.
```

lemonhall, 2017-04-23:
```
124  ps aux | grep blue
  125  sudo service bluetooth stop
  126  sudo hciconfig hci0 up
  127  node index.js
  128  sudo apt-get install libcap2-bin
  129  apt-get autoremove
  130  sudo apt-get autoremove
  131  clear
  132  ls
  133  rm -rf node_modules/
  134  npm install bleno
  135  sudo apt-get install libcap2-bin
  136  sudo setcap cap_net_raw+eip $(eval readlink -f `which node`)
  137  node index.js
```
Slove.....It's because the system legcy node's installtion...

after remove those packages...

All things correct.

sandeepmistry, 2017-04-23:
Closing as resolved, as mentioned above issue was with the dev environment.

closed 2017-04-23 by sandeepmistry

lemonhall, 2017-04-24:
OK，but please add these information in README.md

it‘s realy annoying

sandeepmistry, 2017-04-24:
@lemonhall please open a pull request with the changes you'd like to see in the read me ...