noble/bleno #297: TypeError: Object #<Object> has no method 'emit'

mone741120, 2017-04-26:
When I set my BLE module as the peripheral mode and run the echo example.

If I subscribe the indication and then unsubscribe it, the following error will occur.

root: /home/ubuntu/test/node_modules/bleno/lib/hci-socket/gatt.js:914
            handleAttribute.emit('unsubscribe');
                            ^
TypeError: Object #<Object> has no method 'emit'

pimterry, 2017-05-05:
I can reproduce this too:

```
/usr/src/app/node_modules/bleno/lib/hci-socket/gatt.js:914
            handleAttribute.emit('unsubscribe');
                            ^

TypeError: handleAttribute.emit is not a function
    at Gatt.handleWriteRequestOrCommand (/usr/src/app/node_modules/bleno/lib/hci-socket/gatt.js:914:29)
    at Gatt.handleRequest (/usr/src/app/node_modules/bleno/lib/hci-socket/gatt.js:344:23)
    at Gatt.onAclStreamData (/usr/src/app/node_modules/bleno/lib/hci-socket/gatt.js:274:8)
    at emitTwo (events.js:111:20)
    at AclStream.emit (events.js:194:7)
    at AclStream.push (/usr/src/app/node_modules/bleno/lib/hci-socket/acl-stream.js:26:10)
    at BlenoBindings.onAclDataPkt (/usr/src/app/node_modules/bleno/lib/hci-socket/bindings.js:201:21)
    at emitThree (events.js:116:13)
    at Hci.emit (events.js:197:7)
    at Hci.onSocketData (/usr/src/app/node_modules/bleno/lib/hci-socket/hci.js:456:14)
```

In my case this is reproducible by just connecting, subscribing to 0x1801 and then unsubscribing, with nRF Connect. The unsubscribe immediately throws this error and ends the process for me.

sandeepmistry, 2018-03-12:
@mone741120 @pimterry could you please try out PR #369 and provide you feedback.

pimterry, 2018-03-13:
I can't easily, sorry @sandeepmistry, I'm not working on any bluetooth stuff right now, and I don't have any hardware to hand.

lilerout1590, 2021-04-22:
Excuse me for cutting in, I started using bleno recently.

Same error was occured, in my case too.
Then I added code at line between 914-915 that referring to #369.
But error was not fixed it.
Console says "handleAttribute.emit is not a function".
The action of temporarily disapear error is to comment out statement else block at line 913. I know that doesn't become a solution to the root of the problem...

I'm using:
bleno Version 0.5.0
blueZ Version 5.50
Linux version 4.19.75+

Might be something is missing?