noble/bleno #298: Linux kernel 3.6

gtalusan, 2017-04-27:
I noticed the documentation says Linux kernel 3.6.  What specifically about blend requires 3.6 and above?

DanielTimLee, 2017-04-30:
guess `Bluez` has dependency on Kernel 3.4 or above.

Especially with BLE support, 3.5 is needed

> With this release BlueZ only supports the new Bluetooth Management kernel interface that was introduced in Linux 3.4, so essentially this is the minimum kernel requirement for BlueZ 5. For Low Energy support at least kernel version 3.5 is needed.

http://www.bluez.org/release-of-bluez-5-0/

sandeepmistry, 2017-05-08:
@gtalusan I think this was a requirement because older versions used L2CAP kernel sockets. Now HCI sockets are used, so this may no longer be a requirement.

I don't have a Linux kernel system with older than 3.6 to test, but it would be great if you could test it out and submit a PR to remove the requirement.

closed 2017-05-08 by sandeepmistry