noble/bleno #299: Coordinating requests with multiple devices?

milgner, 2017-04-30:
From what I see, the `onWriteRequest` callback receives a data buffer and an offset, but there doesn't seem to be any indication from which device the write is coming from? So if there is anything other than a static buffer that everyone is reading from, how can the application determine which buffer to continue delivering data from?

Similar for read requests: it doesn't seem to know which device it's currently delivering data to?

In itself it's not a big problem but what confuses me is that BLE central libraries (including your [noble](https://github.com/sandeepmistry/noble) library) seem to be able to read and write reasonably large buffers without specifying offsets. 
What's more, they also seem to know at what point the transmission was complete even if my `bleno`-based `onReadRequest` handler needed to be called 4 times to successfully deliver the whole buffer.

Is there a way for `onWriteRequest` to determine that the whole buffer was transferred successfully from the other side without going to the application level?

milgner, 2017-04-30:
Nevermind... I just read the GATT specification and realized that there can only be one device connected at a time so of course my question is totally bogus...

So at least I can use this opportunity to express my thanks for this fantastic library! I looked at a number of programming languages but bleno/JS seems to be the only viable solution to prototype a peripheral... :+1:

closed 2017-04-30 by milgner

sandeepmistry, 2017-05-08:
@milgner thanks for the positive feedback!

> Nevermind... I just read the GATT specification and realized that there can only be one device connected at a time so of course my question is totally bogus...

This is correct for Bluetooth 4.0 devices. 4.1 and later do allow for more than one connection, but this library does not have support for this currently.

Betatester723, 2019-05-10:
> @milgner thanks for the positive feedback!
>
>
> This is correct for Bluetooth 4.0 devices. 4.1 and later do allow for more than one connection, but this library does not have support for this currently.

Hi Sandeep, 
do you know if multiple connections are not yet supported???