noble/bleno #301: Services undiscoverable in Android?

seanfar, 2017-05-02:
Hi there. Thank you for your work with this package.

I'm currently running into an issue while trying to discover services on Android devices (using nRF connect, to be specific). nRF Control seems to connect to the peripheral just fine, then during a call to `discoverServices()` the result is a big fat 0. Shortly thereafter, the device disconnects. It should be noted that using LightBlue on iOS yields positive results.

I searched through previous issues and couldn't find anything, though it seems someone outlined an issue [here](https://github.com/sandeepmistry/bleno/issues/24#issuecomment-287656835) which may be related (?) It sounds almost identical to what I'm experiencing.

Here are some details with my setup:

- Raspberry Pi Zero W running `Linux 4.9.24+ #993 Wed Apr 26 17:56:54 BST 2017 armv6l GNU/Linux`
- BlueZ version: `5.23`
- Executed `sudo systemctl stop bluetooth` 
- Executed `sudo hciconfig hci0 up` (restarted a few times as well...)

It should also be noted that I am using noble in the same project, though it isn't being utilized at the same time as bleno. Nevertheless, I went ahead and added the environment variable setting 
`NOBLE_MULTI_ROLE=1 `

Here is the relevant hcidump for the complete interaction between devices (on Android):
[hcidump.txt](https://github.com/sandeepmistry/bleno/files/971587/hcidump.txt)

Any and all help is appreciated, and please let me know if any further information is required. Thanks!

sandeepmistry, 2017-05-08:
```

sandeepmistry, 2017-05-08:
```
2017-05-02 16:40:54.123068 > HCI Event: LE Meta Event (0x3e) plen 10
    LE Connection Update Complete
      status 0x00 handle 65
      interval 67.50ms, latency 0.00ms, superv. timeout 20000.00ms
2017-05-02 16:41:23.284044 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 65 reason 0x13
    Reason: Remote User Terminated Connection
```

Strange, there's a disconnect after the connection update complete event. I'll have to try to reproduce this locally ...

sandeepmistry, 2017-05-15:
@seanfar are you using noble and bleno together?

Have you tried just using bleno? What code are you using?

seanfar, 2017-05-15:
@sandeepmistry thanks for taking the time to look into this.

I am using both in the project, I use bleno to setup WiFi over BLE on the pi, then on successfully connecting to a network, the project switches to using noble to scan for proprietary beacons (similar to the Eddystone protocol). I did set `NOBLE_MULTI_ROLE=1` and that did solve an issue I was having, but not this one in particular.

Here is a snippet of some relevant code, I tried to follow your examples as closely as possible:

ble-service.js:
```
var bleno                           = require('bleno');
var PrimaryService                  = bleno.PrimaryService;
var ScanWifiCharacteristic          = require('./characteristics/ScanWifiCharacteristic');
var SetWifiCharacteristic           = require('./characteristics/SetWifiCharacteristic');
var RegisterDeviceCharacteristic    = require('./characteristics/RegisterDeviceCharacteristic');
var wifi                            = require('wifi-control');

var deviceConnected = false;

var primaryService = new PrimaryService({
    uuid: 'd304dbd96fdc4bf3a617e015edd87ed5',
    characteristics: [
      new ScanWifiCharacteristic(),
      new SetWifiCharacteristic(),
      new RegisterDeviceCharacteristic()
    ]
  })

exports.startAdvertising = function() {
  var _this = this;

  wifi.init();
  
  bleno.on('stateChange', function(state) {
    console.log('on -> stateChange: ' + state);
    if (state === 'poweredOn') {
      bleno.startAdvertising('DottedBeacon', [primaryService.uuid]);
    } else {
      bleno.stopAdvertising();
    }
  });

  bleno.on('accept', function(cAddress) { 
    deviceConnected = true;
    console.log('client accepted: ' + cAddress + 'device connected: ' + deviceConnected);
    // console.log(_this);
  });

  bleno.on('disconnect', function(cAddress) { 
    deviceConnected = false;
    console.log('client disconnected: ' + cAddress + 'device connected: ' + deviceConnected);    
  })

  bleno.on('advertisingStart', function(error) {
    console.log('on -> advertisingStart: ' + (error ? 'error ' + error : 'success'));

    if (!error) {
      bleno.setServices([primaryService], function(error){
        console.log('setServices: '  + (error ? 'error ' + error : 'success'));
      });
    } else {
        bleno.stopAdvertising();
        bleno.startAdvertising('DottedBeacon', [primaryService.uuid]);
    }
  });
}

exports.stopAdvertising = function(cb) { 
  bleno.stopAdvertising(cb);
}

exports.getDeviceConnected = function() { return deviceConnected; }

```

And one of my example characteristics,
ScanWifiCharacteristic.js
```
var util                = require('util');
var bleno               = require('bleno');
var BlenoCharacteristic = bleno.Characteristic;
var wifi                = require('wifi-control');
var Descriptor          = bleno.Descriptor;

function ScanWifiCharacteristic() {
  ScanWifiCharacteristic.super_.call(this, {
    uuid: '5099cbc8a71f42928158bf4f25ae9948',
    properties: ['notify'],
    descriptors: [ 
      new Descriptor({
        uuid: 'fff1',
        value: 'Scans for WiFi networks in range and returns the collection.' // static value, must be of type Buffer or string if set
      })
    ]
  });
};

util.inherits(ScanWifiCharacteristic, BlenoCharacteristic);

ScanWifiCharacteristic.prototype.onSubscribe = function(maxValueSize, updateValueCallback) {
    wifi.scanForWiFi(function(err, response) {
      if (err) console.log(err);

      var data = new Buffer(JSON.stringify(response));

      var lastSliceIndex = null;
      var tempBuffer = null;
      // Loop through the data at iterations of the max val size ( + 1 for exclusive slices)
      for(var x = 0; x < data.length; x += maxValueSize)
      {
        tempBuffer = data.slice(x, x + maxValueSize);
        // console.log("BUFF #[${x} - ${x + maxValueSize + 1}]: " + tempBuffer.toString())
        updateValueCallback(tempBuffer)

        // Set this last slice index for potentially sending tail of data
        lastSliceIndex = x + maxValueSize;
      }
      
      if(lastSliceIndex < data.length) {
        tempBuffer = data.slice(lastSliceIndex);
        // console.log("LAST BUFF: " + tempBuffer.toString());
        updateValueCallback(tempBuffer);
      }

      updateValueCallback(new Buffer("EOM"));
    });
};

module.exports = ScanWifiCharacteristic;
```

sandeepmistry, 2017-05-15:
@seanfar ok, I think I know what the problem is. If you look at BLE state machines, for the most part if you start scanning you can no longer advertise and accept connections.

You might need 2 Bluetooth adapters to accomplish your goal ...

seanfar, 2017-05-16:
@sandeepmistry thanks again for your help.

I do realize that you can't be scanning and accepting connections at the same time - the idea was that once the device had a successful wifi connection, it would stop advertising and _then_ begin scanning, so the adapter would only be either advertising OR scanning, in theory.

Removing noble did solve it, so that was the culprit, though it didn't really solve my issue. I made a slight change in the way that my scanner service is initialized (and subsequently, noble) and it seems to be working. Essentially 'lazy loading' my abstracted scanner service did the trick.

Whether it's a good solution or not has yet to be seen, but initial testing has everything working properly. Thanks again, and feel free to close the issue.

sandeepmistry, 2017-05-16:
@seanfar thanks for the update, closing as resolved.

closed 2017-05-16 by sandeepmistry

CBZakaria, 2018-06-27:
@seanfar Thanks for your share

Can you show me your SetWifiCharacteristic too?