noble/bleno #302: connection stuck after disconnection

mone741120, 2017-05-03:
I use echo example to set our BLE module as peripheral and use "nRF Connect" App of Nexus 7 2013 and iPhone 6 to connect the BLE module.

I can rapid connect it successfully firstly, but the 2nd connection will be stuck when I try connect it again after I disconnect the first connection.

There are no any log, just connection stuck unless I use ctrl+c to interrupt and  then reset the "echo example".

sandeepmistry, 2017-05-08:
@mone741120 please provide an HCI dump capture: `sudo hcidump -t -x`

mone741120, 2017-05-08:
Hi, sandeepmistry

my HCI dump is following;

2017-05-08 08:59:00.725050 > HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 1025, role slave
      bdaddr 30:85:A9:DD:FB:84 (Public)
2017-05-08 08:59:00.729066 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr 30:85:A9:DD:FB:84 type 0
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 40 max_interval 56
    latency 0 supervision_to 42 min_ce 0 max_ce 0
2017-05-08 08:59:00.735028 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x0c ncmd 1
    Error: Command Disallowed
2017-05-08 08:59:00.747400 < ACL data: handle 1025 flags 0x00 dlen 7
    ATT: MTU req (0x02)
      client rx mtu 256
2017-05-08 08:59:00.806609 > ACL data: handle 1025 flags 0x02 dlen 11
    ATT: Read By Group req (0x10)
      start 0x0001, end 0xffff
      type-uuid 0x2800
2017-05-08 08:59:00.807972 < ACL data: handle 1025 flags 0x00 dlen 9
    ATT: Error (0x01)
      Error: Request not supported (6)
      Read By Group req (0x10) on handle 0x0000
2017-05-08 08:59:00.822204 < ACL data: handle 1025 flags 0x00 dlen 24
    ATT: Read By Group resp (0x11)
      attr handle 0x0001, end group handle 0x0005
      value 0x00 0x18
      attr handle 0x0006, end group handle 0x0009
      value 0x01 0x18
      attr handle 0x000a, end group handle 0x000d
      value 0x00 0xec
2017-05-08 08:59:00.855394 > ACL data: handle 1025 flags 0x02 dlen 7
    ATT: MTU resp (0x03)
      server rx mtu 256
2017-05-08 08:59:00.859992 < ACL data: handle 1025 flags 0x00 dlen 9
    ATT: Error (0x01)
      Error: Request not supported (6)
      MTU resp (0x03) on handle 0x0000
2017-05-08 08:59:00.952897 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 1025 packets 2
2017-05-08 08:59:01.001642 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 1025 packets 2
2017-05-08 08:59:03.686066 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 1025 reason 0x13
    Reason: Remote User Terminated Connection
2017-05-08 08:59:03.702090 < HCI Command: LE Set Advertising Parameters (0x08|0x0006) plen 15
    min 1280.000ms, max 1280.000ms
    type 0x03 (ADV_NONCONN_IND - Non connectable undirected advertising) ownbdaddr 0x00 (Public)
    directbdaddr 0x00 (Public) 00:00:00:00:00:00
    channelmap 0x07 filterpolicy 0x00 (Allow scan from any, connection from any)
2017-05-08 08:59:03.702578 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertising Parameters (0x08|0x0006) ncmd 1
    status 0x00
2017-05-08 08:59:03.702623 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  01
2017-05-08 08:59:03.703173 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x00
2017-05-08 08:59:03.734390 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  01
2017-05-08 08:59:03.735236 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x0c
    Error: Command Disallowed
2017-05-08 08:59:38.099425 > HCI Event: LE Meta Event (0x3e) plen 30
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr CC:A1:F5:53:4D:EF (Random)
      Unknown type 0x07 with 16 bytes data
      RSSI: -81
2017-05-08 08:59:38.249325 > HCI Event: LE Meta Event (0x3e) plen 42
    LE Advertising Report
      ADV_NONCONN_IND - Non connectable undirected advertising (3)
      bdaddr F9:BC:74:04:0E:EA (Random)
      Flags: 0x06
      Unknown type 0xff with 25 bytes data
      RSSI: -97
2017-05-08 08:59:39.296421 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2017-05-08 08:59:39.297248 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00

sandeepmistry, 2017-05-15:
@mone741120 what code are you using? Is something triggering a scan after the disconnect?

```
2017-05-08 08:59:38.099425 > HCI Event: LE Meta Event (0x3e) plen 30
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr CC:A1:F5:53:4D:EF (Random)
      Unknown type 0x07 with 16 bytes data
      RSSI: -81
2017-05-08 08:59:38.249325 > HCI Event: LE Meta Event (0x3e) plen 42
    LE Advertising Report
      ADV_NONCONN_IND - Non connectable undirected advertising (3)
      bdaddr F9:BC:74:04:0E:EA (Random)
      Flags: 0x06
      Unknown type 0xff with 25 bytes data
      RSSI: -97
2017-05-08 08:59:39.296421 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2017-05-08 08:59:39.297248 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
```

mone741120, 2017-05-16:
@sandeepmistry 
Actually, when I disconnect, the log stops at

2017-05-08 08:59:03.735236 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x0c
    Error: Command Disallowed

and then I can not reconnect.

The following contents you grabbed from my log is generated by other operations. Maybe I restart peripheral mode.