noble/bleno #303: Using Noble & Bleno, Noble's Peripheral Connection event emits Bleno's accept event.

DanielTimLee, 2017-05-08:
Using Noble & Bleno, 
`Noble's Peripheral Connection event` emits Bleno's accept event.

I think this is quite wrong circumstance. I'm not sure that I'm the only one having this problem.

Tested on `Macbook 10.12 Sierra (early 2015) Broadcom 20703A1`

sandeepmistry, 2017-05-15:
@DanielTimLee please provide a code sample to reproduce this, along with steps to reproduce.

DanielTimLee, 2017-05-20:
Sorry for my late sample code update!

I have a nRF51-dk device advertising with `00009000000000108000001122334455` this uuid service (128bit-vendor uuid).

```js
var noble = require('noble');
var bleno = require('bleno');

bleno.on('accept', () => console.log("HI! I'm Peripheral's Accept Event!"));

noble.on('stateChange', function (state) {
  if (state === 'poweredOn') {
    console.log('scanning...');
    noble.startScanning(['00009000000000108000001122334455']);
  }
  else {
    noble.stopScanning();
  }
});

noble.on('discover', dev => {
  noble.stopScanning();

  dev.connect((e) => console.log('result:' + ((e) ? e : 'success')));
});
```

and the result goes like this.

> scanning...
result:success
HI! I'm Peripheral's Accept Event!

I think this situation might be pretty wrong.
Or am I just misunderstanding the meaning of `Accept`?