noble/bleno #307: Nodejs doesn't react at all

arizona2014, 2017-05-09:
I'm trying to follow this tutorial -  https://www.sitepoint.com/communicating-with-bluetooth-low-energy-devices-in-cordova/ and it's weird what I'm gonna say next but mainly this is my problem : Due to successful data submition from IONIC2 app to Nodejs back ... i dont see anything logged on the console how it should.

So my confirguration is as follows : 
IONIC2 / BLE lib  for the frontend and
PI Zero W / NodeJS 6 / NPM 3 for backend

Running the command **sudo node attendees.js** will console log me "started... "  as it should do. But when I submit some datas from IONIC 2 to the connected device ... I see no reaction ( error /success ) from the running server.

Any ideeas why is that and how can I debug ?

arizona2014, 2017-05-12:
Couldn't let this error like that so I downloaded BLE Scanner on my android, discovered quickly my PI's ble connection and after connection, got for the first time same 3 services as my IONIC2 app.

Fortunatelly I've unpaired my test device and connected again, and voila ... this time I get the good characteristics : ffffffff-ffff-ffff-ffff-fffffffffff1, ffffffff-ffff-ffff-ffff-fffffffffff2 ... ffffffff-ffff-ffff-ffff-fffffffffff6

I subscribe on ffffffff-ffff-ffff-ffff-fffffffffff5 characteristic and by magic my nodejs app comes alive and starts incrementing each 5 seconds ... great news. So now I have to replicate this on my IONIC2 app. I now i connect successfully with test.js, also I know the nodejs does well its incrementing so i'm on the good path.

I have some questions though : 

* at this moment test.js accepts connections and sends notifications from/to any client. Is there a way to connect / send connection only to my IONIC2 app ? How can I implement that and where ?

* for the moment test.js sends only an integer value corresponding to a incremention. Can I send a file to my app ?

sandeepmistry, 2017-05-14:
@arizona2014 

> at this moment test.js accepts connections and sends notifications from/to any client. Is there a way to connect / send connection only to my IONIC2 app ? How can I implement that and where ?

Nope

> or the moment test.js sends only an integer value corresponding to a incremention. Can I send a file to my app ?

You can only send up to a `maxValueSize` byte per notification.

closed 2017-05-14 by sandeepmistry