noble/bleno #310: UART equivalent

anweiss, 2017-05-12:
I think I'm asking the right question, but is there a UART implementation equivalent to https://github.com/sandeepmistry/arduino-BLEPeripheral/tree/master/examples/serial in bleno? Would love to see this in the examples.

sandeepmistry, 2017-05-15:
I'm not too keen on UART's over BLE in general, they generally don't leverage the benefits of GATT.

@anweiss What were you thinking the example would do? How do you think this would be beneficial to the community?

@jacobrosenthal any thoughts on this.

anweiss, 2017-05-15:
So I ended up finding this -> https://github.com/xykox/nrfUART-echo which is what I was looking for.

sandeepmistry, 2017-05-15:
@anweiss so can we close this?

anweiss, 2017-05-15:
Fine by me. Thanks

closed 2017-05-15 by anweiss