noble/bleno #311: how to solve (android) disconnect after 30 sec

Dabin22, 2017-05-15:
onClientConnectionState() - status=22 -> (android log) local host terminate

only one characteristic used -> property : notify and  onSubscribe

koustabhdolui, 2017-05-19:
I am having the same problem. I tried an app called BLE Scanner and I also wrote a small app myself using my phone as a central device, in both cases the phone disconnects after 30 seconds. Did you find a solution @Dabin22

koustabhdolui, 2017-05-19:
Hi @Dabin22. I managed to find why it was happening in my case. I was using a BLE dongle on the Pi and turning it up while turning the internal UART interface down.

hci1:	Type: BR/EDR  Bus: UART
	BD Address: B8:27:EB:38:88:28  ACL MTU: 1021:8  SCO MTU: 64:1
	DOWN 
	RX bytes:4100 acl:80 sco:0 events:243 errors:0
	TX bytes:5822 acl:79 sco:0 commands:170 errors:0

hci0:	Type: BR/EDR  Bus: USB
	BD Address: 00:1A:7D:DA:71:04  ACL MTU: 310:10  SCO MTU: 64:8
	UP RUNNING 
	RX bytes:2207 acl:6 sco:0 events:147 errors:0
	TX bytes:4303 acl:11 sco:0 commands:132 errors:0

In this case, the 30 second disconnection was occuring. But when I turned up the hci1 interface and turned down hci0, it worked perfectly for me!

hci1:	Type: BR/EDR  Bus: UART
	BD Address: B8:27:EB:38:88:28  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:4789 acl:80 sco:0 events:281 errors:0
	TX bytes:6576 acl:79 sco:0 commands:208 errors:0

hci0:	Type: BR/EDR  Bus: USB
	BD Address: 00:1A:7D:DA:71:04  ACL MTU: 310:10  SCO MTU: 64:8
	DOWN 
	RX bytes:2207 acl:6 sco:0 events:147 errors:0
	TX bytes:4303 acl:11 sco:0 commands:132 errors:0

Dabin22, 2017-05-23:
Thanks @koustabhdolui 
I had a different problem. We only set the property to notify. It works well with the write setting.
Have a nice day!

sandeepmistry, 2017-05-29:
Closing, as it seemed solved above.

closed 2017-05-29 by sandeepmistry