noble/bleno #314: Disconnect after 30 second

rymca, 2017-05-25:
Hi, i use bleno with LightBlue for testing;
After connection, the device is disconnected after 30 seconds, all time.
I user ble dongle.
I can not understand the reason.
Can you help me?
Thank you

sandeepmistry, 2017-05-29:
@rymca what kernel version and BlueZ `blued` are you using?

rymca, 2017-05-29:
Hi! i develop on MaxOSX (Sierra)

Kernel Version:
      System Version: macOS 10.12.4 (16E195)
      Kernel Version: Darwin 16.5.0

How can i get blueb version?
Thanks1

sandeepmistry, 2017-06-18:
Ah, if it's macOS my previous comments don't apply. Could you please try rebooting both the iOS and macOS device to see if the issue still persists. Sometimes `blued` can get into a confused state.

sandeepmistry, 2018-01-29:
Closing due to lack of activity.

closed 2018-01-29 by sandeepmistry