noble/bleno #316: Bleno segmentation fault in Docker container

cryptax, 2017-05-30:
I have a simple Docker container for Bleno, but when I run `node main.js` with a _basic_ main.js such as 'var bleno = require('bleno');' it core dumps...

```
# node
> require('bleno');
Segmentation fault (core dumped)
```

Of course, it used to work ;)
Any idea?

To reproduce the issue, I provide my Dockerfile & commands. (of course, docker is required).
This is my Dockerfile:
```
FROM node:latest

# Install bleno and dependencies ---------------
# bluetooth IS required despite we do service bluetooth stop in the end
RUN DEBIAN_FRONTEND=noninteractive apt-get update && \
    apt-get install -yqq \
    build-essential \
    bluetooth \
    bluez \
    emacs \
    libbluetooth-dev \
    libudev-dev \
    usbutils \
    && rm -rf /var/lib/apt/lists/*		   

RUN npm install bleno 

WORKDIR /root

# Cleaning up and final setup -------------------------
RUN apt-get autoremove -yqq
RUN apt-get clean

RUN service bluetooth stop
```
To build it: `docker build -t cryptax/bleno:test .`
To run it: `docker -it --rm --privileged cryptax/bleno:test /bin/bash` and then in the shell:

```
# node
> require('bleno');
```

sandeepmistry, 2017-06-18:
@cryptax could you please try again with master? `npm install sandeepmistry/bleno`

This might be address with some changes that have not been published to npm yet.

cryptax, 2017-06-19:
okay - I'll try asap and let you know. Thanks!

cryptax, 2017-06-19:
It's fixed : you need to use **both** `RUN npm install sandeepmistry/bleno` and `FROM node:latest` 
Thanks!

FYI, here is a working Dockerfile:

```
FROM node:latest

MAINTAINER @cryptax

# docker run -it --net=host --privileged --name bleno cryptax/bleno /bin/bash

# Install bleno and dependencies ---------------
# bluetooth IS required despite we do service bluetooth stop in the end
RUN DEBIAN_FRONTEND=noninteractive apt-get update && \
    apt-get install -yqq \
    build-essential \
    bluetooth \
    bluez \
    emacs \
    libbluetooth-dev \
    libudev-dev \
    usbutils \
    && rm -rf /var/lib/apt/lists/*		   

RUN npm install sandeepmistry/bleno
WORKDIR /root

# Cleaning up and final setup -------------------------
RUN apt-get autoremove -yqq
RUN apt-get clean

RUN service bluetooth stop
```

closed 2017-06-19 by cryptax

lukkigi, 2017-06-22:
I'm getting the same error now

```
image: resin/raspberrypi3-node:latest

before_script:
  - apt update
  - apt install bluetooth bluez libbluetooth-dev libudev-dev usbutils emacs build-essential
  - npm --version
  - npm install sandeepmistry/bleno

cache:
  paths:
    - node_modules/

publish:
  stage: deploy
  artifacts:
    name: "$CI_COMMIT_REF_NAME"
  only:
    - master
  script:
    - npm start
```

```
root@939ffbae9019:/# node
> require('bleno')
Segmentation fault (core dumped)
```

zakmckracken1988, 2017-08-10:
This does not seem to be fixed yet. Using sandeepmistry/bleno made things better but did not resolve the issue. With latest node version (8.2.1) things got worse again. Any chance you work on a fix for node 6.11.2 ?

gtrifonov, 2018-04-26:
running docker in network host mode solved issue for me.
sudo docker run  --privileged --rm -it --network host  --name noble  noble