noble/bleno #318: Sending a string a characteristic value

cryptax, 2017-06-19:
Hello,

Would you know how to have a characteristic return a **text string**?

This is the function registered for onReadRequest for my characteristic. 
Value would typically be 'Hello World'.

The following is not working correctly and I haven't found the same in your examples. Can you help?

```
function myRead(offset, callback) {
    console.log('[+] Sending back: ', new Buffer(value));
    callback(this.RESULT_SUCCESS, new Buffer(value));
}
```

Thanks

cryptax, 2017-06-20:
I found the solution myself in the code test.js. You need to slice data, or you get multiple times the buffer.

```javascript
function myRead(offset, callback) {
    var result = this.RESULT_SUCCESS;
    var data = new Buffer(value);

    console.log('[+] Sending back: ', data);
    if (offset > data.length) {
	result = this.RESULT_INVALID_OFFSET;
	data = null;
    } else {
	data = data.slice(offset);
    }
    callback(result, data);
}
```

closed 2017-06-20 by cryptax

narek-king, 2017-06-21:
Thank you for the solution. It helped me.