noble/bleno #320: Emulate a HID device

neophob, 2017-06-23:
Hi, thanks for this library

I try to run bleno to emulate a HID device. My BLE device is a realtek device, so the maxMtu is limited to 23 bytes.

the reportmap (2A4B) characteristic is about 60 bytes, this means a read requst must deliver multiple results - as i see multiple read requests with an increasing offset. my question: what should the callback look like, when the offset exceeds the data (aka end of data)?

just an empty RESULT_SUCCESS?

sandeepmistry, 2017-07-30:
@neophob please see: https://github.com/sandeepmistry/bleno/blob/master/test.js#L59-L75

closed 2017-07-30 by sandeepmistry

wind-rider, 2017-08-12:
@neophob did you succeed to emulate a HID device? Could you please show us how you did that?

willwade, 2017-10-02:
@neophob - I'm with @wind-rider on this one. Even if you're not happy sharing code I Would love to know if you even succeeded. Thanks.