noble/bleno #324: Truncated message with Raspberry Pi

superlopuh, 2017-06-29:
I wrote a script that captures messages from MQTT, and forwards them via Bluetooth to an iPhone. While this worked well on the Mac, when I ran it on a Raspberry Pi, the resulting Buffer of around 240 bytes was truncated, resulting in an invalid JSON.

Since the code is the same, I doubt the bug is in bleno, but I'm wondering whether you know whether I can change something in the Bleno script to make this work.

sandeepmistry, 2017-07-31:
@superlopuh please provide more details, including a simplified Node.js script to reproduce this issue. It's most likely MTU related ...

sandeepmistry, 2018-01-29:
Closing due to lack of activity.

closed 2018-01-29 by sandeepmistry

superlopuh, 2018-01-31:
Hi, I'm no longer working on this, but the code to do it all is in here:
https://github.com/snipsco/snips-mqtt-relay/blob/master/lib/ble.js
I no longer have access to the swift code.