noble/bleno #329: Connection Issue

btrmateusz, 2017-07-19:
A few seconds after connecting, device disconnects. Services are not discovered. I'm using the nRF Connect Android app. My bleno app doesn't do much - just trying to get something simple to work.

Bleno is running on macOS Sierra 10.12.5 - MacBook Pro 2017, with Xcode 9.0 beta 3 with a Swift 4.0 Snapshot 2017-07-06 toolchain.

nRF Connect is running on Android 7.1.1 - Nexus 6.

Tried to reboot the system, kill the blued service, did not help.

Logs below. Hopefully nothing private in those. Anything out of the ordinary?
[Android adb logcat.txt](https://github.com/sandeepmistry/bleno/files/1159252/Android.adb.logcat.txt)
[Android nRF Log.txt](https://github.com/sandeepmistry/bleno/files/1159253/Android.nRF.Log.txt)
[Android Bluetooth HCI snoop log (Google Drive)](https://drive.google.com/open?id=0BzTOhgVpyAq4TUZ4eFgtNF9WWG8)
[MacOS bleno debug.txt](https://github.com/sandeepmistry/bleno/files/1159254/MacOS.bleno.debug.txt)
[MacOS console.txt](https://github.com/sandeepmistry/bleno/files/1159255/MacOS.console.txt)
[MacOS index.js (Google Drive)](https://drive.google.com/open?id=0BzTOhgVpyAq4ekppS2JHYVh4Nms)

GitHub doesn't like some file types so I uploaded them to Google Drive.

sandeepmistry, 2017-07-31:
@btrmateusz could you please also provide your bleno code? Have you tried one of the built-in examples?

btrmateusz, 2017-08-09:
Thanks for getting back to me @sandeepmistry. It looks like it's my phone - I got a hold of two other BLE devices - Samsung S7 and iPad Air 2. Both seem to be able to connect, read and write to characteristics without any disconnects.

My bleno code is found in the ```MacOS index.js (Google Drive)``` file in my initial post.

I used nRF Connect to connect to the bleno peripheral on both Android phones. On the iPad I used LightBlue app, as suggested [here](https://github.com/sandeepmistry/bleno#useful-toolslinks).

Eliminating any JS issues, can you think of any reasons why my Nexus 6 wouldn't work? I am using a stock Google OTA Android image.

btrmateusz, 2017-08-09:
I found [this](https://devzone.nordicsemi.com/question/71121/connection-terminated-under-android-6-after-10-seconds/) thread out there. In my [MacOS bleno debug.txt](https://github.com/sandeepmistry/bleno/files/1159254/MacOS.bleno.debug.txt) file there's this: ```"kCBMsgArgConnectionInterval": 7,```. Is this 7 milliseconds? Is there any way to change this value with XPC?

https://github.com/sandeepmistry/bleno/issues/235

btrmateusz, 2017-08-10:
Got rid of files on Google drive - Github allows *.zip so I zipped up the JS and Android snoop log.

btrmateusz, 2017-08-17:
Just to make sure I have that covered - I flashed my phone to the latest Android OTA image. Same Android version (7.1.1) but a different build with different (latest) security patches. Same issue.

btrmateusz, 2017-08-31:
I'm going to close this - I've been having other BLE issues with this phone (ex., I can't pair using a passkey) - which don't seem to happen on other Android and iOS devices. This phone is EOL and unless other devices have similar problems, it doesn't matter.

closed 2017-08-31 by btrmateusz