noble/bleno #330: Bleno peripheral automatically unsubscribe central  when Bluetooth HFP profile reject the call.

anshadr0405, 2017-07-31:
I am developing an application, which uses Raspberry Pi 2 and Bluetooth Dongle (CSR 4.0) to transfer data from Pi2 to iOS device. We are using both HFP and BLE profile, where Bleno is used for data transfer. Bleno peripheral  is successfully sending notifications to my BLE central till an incoming call comes to my central device (iphone). When an incoming call is recieved on Central (iPhone), Peripheral automatically gets unsubscribed and we cant subscribe again to the connected characteristics.

sandeepmistry, 2017-09-04:
@anshadr0405 please provide debug logs `sudo DEBUG=* node <file>.js` and HCI dump capture `sudo hcidump -t -x`.

I suspect the BLE connection is disconnected.

sandeepmistry, 2018-01-29:
Closing due to lack of activity.

closed 2018-01-29 by sandeepmistry