noble/bleno #333: Calling bleno.disconnect() disconnects device but blocks new connection for another 20secs

mkatter, 2017-08-18:
Hi,
I want to diconnect an inactive device, therefore I am calling bleno.disconnect().
The device gets disconnected - this is the expected behaviour but the device is not able to reconnect for another 20secs.
hcidump -t -x outputs after bleno.disconnect(): 
```
2017-08-18 23:23:05.764200 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 64 reason 0x13
    Reason: Remote User Terminated Connection
2017-08-18 23:23:05.767576 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2017-08-18 23:23:25.708081 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x22
    Reason: LMP Response Timeout
2017-08-18 23:23:25.796586 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  01
2017-08-18 23:23:25.800573 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x00
```

hcidump -t -x outputs this using nRF Connect:
```
2017-08-18 23:27:54.794896 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x13
    Reason: Remote User Terminated Connection
2017-08-18 23:27:54.866602 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  01
2017-08-18 23:27:54.869356 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x00
```

Thanks for your feedback!

sandeepmistry, 2017-09-05:
@mkatter what Bluetooth adapter are you using? It seems to be the culprit, as it takes 20 seconds to perform the command:
```
2017-08-18 23:23:05.764200 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 64 reason 0x13
    Reason: Remote User Terminated Connection
2017-08-18 23:23:05.767576 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2017-08-18 23:23:25.708081 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x22
    Reason: LMP Response Timeout
```

mkatter, 2017-09-21:
@sandeepmistry I am using Nano Pi's built-in bluetooth! So, no external (USB) adapter.
I am wondering about the `Reason: LMP Response Timeout`

rzr, 2021-02-20:
Please confirm this issue with @abandonware version 

Relate-to: https://github.com/abandonware/bleno/issues/17