noble/bleno #337: Feature request, custom manufacturer packet in iBeacon

mlfarrell, 2017-09-14:
See comment in code below

```
Gap.prototype.startAdvertisingIBeacon = function(data) {
  debug('startAdvertisingIBeacon: data = ' + data.toString('hex'));

  var dataLength = data.length;
  var manufacturerDataLength = 4 + dataLength;
  var advertisementDataLength = 5 + manufacturerDataLength;
  var scanDataLength = 0;

  var advertisementData = new Buffer(advertisementDataLength);
  var scanData = new Buffer(0);

  // flags
  advertisementData.writeUInt8(2, 0);
  advertisementData.writeUInt8(0x01, 1);
  advertisementData.writeUInt8(0x06, 2);

  advertisementData.writeUInt8(manufacturerDataLength + 1, 3);
  advertisementData.writeUInt8(0xff, 4);

  ///THIS SHOULD BE CONFIGURABLE
  advertisementData.writeUInt16LE(0x004c, 5); // Apple Company Identifier LE (16 bit)

  advertisementData.writeUInt8(0x02, 7); // type, 2 => iBeacon
  advertisementData.writeUInt8(dataLength, 8);

  data.copy(advertisementData, 9);

  this.startAdvertisingWithEIRData(advertisementData, scanData);
};
```

mlfarrell, 2017-09-14:
Ugh nevermind, hard coding that to something else makes the beacon stop working entirely.

Is there ANY way I can advertise ANY kind of data here that can be detected on the iOS end?  besides just that "its a beacon".  I need to be able to scan for this beacon, identify it, and actually connect to its services

mlfarrell, 2017-09-18:
Okay, so I got it to work with the local name key, see code below.  Adding this to bleno would be quite useful.

```
Gap.prototype.startAdvertisingIBeacon = function(data, localName) {
  debug('startAdvertisingIBeacon: data = ' + data.toString('hex'));

  var dataLength = data.length;
  var manufacturerDataLength = 4 + dataLength;
  var advertisementDataLength = 5 + manufacturerDataLength;
  var scanDataLength = 0;

  var advertisementData = new Buffer(advertisementDataLength);
  var scanData = new Buffer(0);

  // flags
  advertisementData.writeUInt8(2, 0);
  advertisementData.writeUInt8(0x01, 1);
  advertisementData.writeUInt8(0x06, 2);

  advertisementData.writeUInt8(manufacturerDataLength + 1, 3);
  advertisementData.writeUInt8(0xff, 4);

  advertisementData.writeUInt16LE(0x004c, 5); // Apple Company Identifier LE (16 bit)

  advertisementData.writeUInt8(0x02, 7); // type, 2 => iBeacon
  advertisementData.writeUInt8(dataLength, 8);
  
  //allow the setting of kCBAdvDataLocalName key from argument  - mlf
  if(localName !== undefined)
  {
    var name = localName;
    var scanDataLength = 0;
    
    if (name && name.length) {
      scanDataLength += 2 + name.length;
    }
    
    var scanData = new Buffer(scanDataLength);
    var nameBuffer = new Buffer(name);

    scanData.writeUInt8(1 + nameBuffer.length, 0);
    scanData.writeUInt8(0x08, 1);
    nameBuffer.copy(scanData, 2);    
  }
  ///////////////////////////////////////////////////////////////////////

  data.copy(advertisementData, 9);

  this.startAdvertisingWithEIRData(advertisementData, scanData);
};
```

closed 2018-01-29 by sandeepmistry

bengotow, 2018-08-24:
Thank you! This code is exactly what I needed 👍  Definitely seems like a high level API for this would be useful.