noble/bleno #341: cannot run on HighSierra

lkknguyen, 2017-10-09:
I was trying to run Bleno on Mac HighSierra but it doesn't work. I was working before with Sierra. No services shown on LightBlue.

`default	21:43:00.463494 +0200	identityservicesd	Continuity stop advertising of type: 12
default	21:43:00.465117 +0200	wirelessproxd	Advert timer expired for client 12 after 2.00 seconds
default	21:43:00.466371 +0200	wirelessproxd	Stop advertising for process identityservicesd (811) of type 12
default	21:43:00.466473 +0200	identityservicesd	Continuity update advertising with data: Advertising request of type 12, priority 4, UseFG 290 (181.25 ms), data <003011f9 1fa70586 c5df78ca ccb1>, connectable 1, options {
}
default	21:43:00.466451 +0200	wirelessproxd	Nothing left to advertise, stopping
default	21:43:00.468017 +0200	wirelessproxd	Tried to update the advertising request interval for process identityservicesd (811), but a client has asked us to stop in the meantime
default	21:43:00.471081 +0200	kernel	127403.486887 Bluetooth USB Host Controller@14330000: IOUSBHostDevice::forcePowerGated: kPowerStateOn timeout 0ms
default	21:43:00.536149 +0200	kernel	127403.551949 Bluetooth USB Host Controller@14330000: IOUSBHostDevice::powerStateWillChangeToGated: kPowerStateOn
default	21:43:00.536218 +0200	kernel	127403.552023 Bluetooth USB Host Controller@14330000: IOUSBHostDevice::setPowerStateGated: kPowerStateSuspended -> kPowerStateOn
default	21:43:00.536261 +0200	kernel	127403.552070 Bluetooth USB Host Controller@14330000: IOUSBHostDevice::setPowerStateGated: resuming pipes
default	21:43:00.536318 +0200	kernel	127403.552119 Bluetooth USB Host Controller@14330000: IOUSBHostDevice::forcePowerGated: kPowerStateOn completed with 0x00000000 after 65ms
default	21:43:00.536331 +0200	kernel	127403.552140 Bluetooth USB Host Controller@14330000: IOUSBHostDevice::powerChangeDone:
default	21:43:00.536395 +0200	kernel	127403.552198 Bluetooth USB Host Controller@14330000: IOUSBHostDevice::updateIdlePolicyGated: IOUSBHostInterface@2 [<private>] has an idle policy of 500ms
default	21:43:00.536416 +0200	kernel	127403.552222 Bluetooth USB Host Controller@14330000: IOUSBHostDevice::updateIdlePolicyGated: IOUSBHostInterface@0 [<private>] has an idle policy of 500ms
default	21:43:00.536436 +0200	kernel	127403.552242 Bluetooth USB Host Controller@14330000: IOUSBHostDevice::updateIdlePolicyGated: IOUSBHostInterface@1 [<private>] has an idle policy of 500ms
default	21:43:00.536454 +0200	kernel	127403.552263 Bluetooth USB Host Controller@14330000: IOUSBHostDevice::updateIdlePolicyGated: consensus idle policy timeout 500ms
default	21:43:01.578633 +0200	kernel	127404.594417 Bluetooth USB Host Controller@14330000: IOUSBHostDevice::forcePowerGated: kPowerStateSuspended timeout 1000ms
default	21:43:01.578924 +0200	kernel	127404.594713 Bluetooth USB Host Controller@14330000: IOUSBHostDevice::powerStateWillChangeToGated: kPowerStateSuspended
default	21:43:01.578971 +0200	kernel	127404.594759 Bluetooth USB Host Controller@14330000: IOUSBHostDevice::setPowerStateGated: kPowerStateOn -> kPowerStateSuspended
default	21:43:01.578982 +0200	kernel	127404.594773 Bluetooth USB Host Controller@14330000: IOUSBHostDevice::setPowerStateGated: suspending pipes
default	21:43:01.579083 +0200	kernel	127404.594872 Bluetooth USB Host Controller@14330000: IOUSBHostDevice::forcePowerGated: allowing power level to float
default	21:43:01.579100 +0200	kernel	127404.594891 Bluetooth USB Host Controller@14330000: IOUSBHostDevice::forcePowerGated: kPowerStateSuspended completed with 0x00000000 after 0ms
default	21:43:09.948053 +0200	system_profiler	Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
default	21:43:09.949102 +0200	system_profiler	Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
default	21:43:09.949771 +0200	system_profiler	Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
default	21:43:09.950400 +0200	system_profiler	Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
default	21:43:09.951137 +0200	system_profiler	Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
default	21:43:09.951666 +0200	system_profiler	Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
default	21:43:09.952334 +0200	system_profiler	Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
default	21:43:09.952742 +0200	system_profiler	Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
default	21:43:09.953535 +0200	system_profiler	Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
default	21:43:09.955304 +0200	system_profiler	Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
default	21:43:09.956218 +0200	system_profiler	Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
default	21:43:09.957073 +0200	system_profiler	Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
default	21:43:09.957863 +0200	system_profiler	Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
default	21:43:09.958966 +0200	system_profiler	Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
default	21:43:09.959800 +0200	system_profiler	Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
default	21:43:09.961419 +0200	system_profiler	Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
default	21:43:09.962761 +0200	system_profiler	Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
default	21:43:09.964383 +0200	system_profiler	Exception NSInvalidArgumentException thrown while decoding IOBluetoothSDPServiceRecord
default	21:43:10.322949 +0200	bluetoothd	setStateForConnection CBCentralManagerStatePoweredOn
`

nraamat, 2017-10-09:
I see the exact same!

nraamat, 2017-10-10:
In mac/bindings.js it seems to try to talk to "com.apple.blued" through XPC:

`  this._xpcConnection = new XpcConnection('com.apple.blued');`

But High Sierra has no running services called "com.apple.blued"

`sudo launchctl list | grep bl`
`           0       com.apple.bluetoothReporter`
`           0       com.apple.diagnosticextensions.osx.bluetooth.helper`
`108     0       com.apple.bluetoothd`
`           0       com.apple.bluetoothaudiod`

nraamat, 2017-10-10:
I changed: 

`this._xpcConnection = new XpcConnection('com.apple.blued');`

to this

`this._xpcConnection = new XpcConnection('com.apple.blued');`

and this

`107 blenoBindings.on('kCBMsgId4', function(args) {`

to this

`107 blenoBindings.on('kCBMsgId6', function(args) {`

in mac/bindings.js and it seems to work

lkknguyen, 2017-10-10:
Great thx @nraamat , possible to make a PR?

nraamat, 2017-10-10:
I might not have the time for the PR - also I am not sure if this is a stable solution

nraamat, 2017-10-10:
I experience that somethings work and some don't. Probably the name of the bluetooth deamon was not the only thing that changed.

Does anybody know the where to find documentation for XPC to com.apple.bluetoothd

lkknguyen, 2017-10-10:
Does this help? https://github.com/sandeepmistry/noble/pull/689/commits

nraamat, 2017-10-10:
It is heavily related - I just don't know how to get the reference/documentation for needed constants for high sierra

CocoaBob, 2017-10-11:
@malikolivier 
After installing your modified fork, it still doesn't work. `bleno.on('stateChange', function (state)` is never called, I'm using macOS 10.13. But I'm very new to Node.JS, I'm not sure if it's correct to use `npm install ./` in the downloaded `bleno`'s folder to install the modified local package?
Thanks in advance!

malikolivier, 2017-10-11:
@CocoaBob My fork is not modified. It was actually merged into the main trunk and my contribution was released in 0.4.2. See https://github.com/sandeepmistry/bleno/pull/274

What makes you assume that my fork might work for your specific use case?

CocoaBob, 2017-10-11:
@malikolivier Sorry,sorry, sorry! I was trying to mention @lkknguyen but I typed `ikk`, I thought it was `I`, then GitHub automatically selected you 😂

I was trying to make it work under High Sierra, and @lkknguyen 's fork is supposed contain the fix, but it doesn't work on my side.

lkknguyen, 2017-10-11:
It didn’t work for me yet, sorry couldn’t help you. I was just trying out what nraamat suggested.

nraamat, 2017-10-11:
Yes, an unfortunately it was only fixing the first of many problems. And I am not able to do the necessary reverse engineering of CoreBluetooth.

aj-ptw, 2017-10-13:
This actually seems pretty real. Did anyone get this working?

aj-ptw, 2017-10-13:
@lkknguyen try the pr i just submitted #343

lkknguyen, 2017-10-17:
@aj-ptw Thanks, still unable to make it work. Is it working with your MacOS now?

aj-ptw, 2017-10-17:
@lkknguyen nope. still needs more work i think.

lkknguyen, 2017-11-01:
Have anyone got further insights to this?

maurobender, 2017-11-02:
I can report the same issue in MacOs High Sierra 10.13.

Terfender, 2017-12-13:
For me on MacOs HighSierra 10.13.1, I changed `blenoBindings.on('kCBMsgId6', ` to `blenoBindings.on('kCBMsgId4', ` in `lib/mac/bindings.js` file

aj-ptw, 2017-12-13:
@Terfender in which file?

Terfender, 2017-12-13:
@aj-ptw mentioned in  `lib/mac/bindings.js` file. I believe [this pull request](https://github.com/sandeepmistry/noble/pull/689/files#diff-81c793ee25a4a5820d33c0d4c9beeea0) will solve the issue

aj-ptw, 2017-12-13:
@Terfender will changing the lib/mac/bindings.js break the previous version? 

So you didnt need to make a new file? Could you PR to this repo with your fix so I can test it!

How did you know to change from a 6 to a 4?

Thanks!

Terfender, 2017-12-13:
@aj-ptw , I am kinda naive when it comes to create PR on github, I created a new branch. [PR](https://github.com/sandeepmistry/bleno/pull/357) here, and my mac can pickup stateChange now. However, it doesn't pickup `advertisingStart` event. Looks like `kCBMsgId16` needs to change? I have no idea what these IDs are, but I will investigate and see if I come up with something.

I knew about changeing the ID from 6 to 4 from [this comment](https://github.com/sandeepmistry/bleno/issues/341#issuecomment-335374232).

jacobrosenthal, 2017-12-25:
Please test https://github.com/sandeepmistry/bleno/pull/360

sandeepmistry, 2018-01-23:
Closed via PR #360 from @jacobrosenthal.

You can test master via `npm install sandeepmistry/bleno`

closed 2018-01-23 by sandeepmistry