noble/bleno #342: systemd service does not start advertising after boot but restarting the service works.

ImJCabus, 2017-10-12:
Some key data:
- Raspberry Pi 3
- Ubuntu Mate 15.04

I'm trying to auto start and keep alive my Node.js app featuring bleno.
However, something's weird with systemd and bleno behaviour:

This is my systemd service located at `/lib/systemd/system`:

[image]

`index.js`:

[image]

Right after `sudo reboot`, `sudo systemctl status mx-ble-node.service` shows this:

[image]

Note that `bleno.state == powerdOff`.
bleno is not advertising. `onStateChange` is never called.

However, if I `sudo systemctl restart mx-ble-node.service`, I get the following:

[image]

Note that this time, `bleno.state == 'unknown'`.
`onStateChange` is called and the service is advertised correctly.

Things I've checked so far:
- Service is running (enabled)
- No difference if I start my service `After=` or `Before=` `bluetooth`.
- No difference if `bluetooth` is enabled or disabled. (bleno README suggests disabling it)

Why isn't it working after boot, but after manual service restart?

arget888, 2017-11-18:
Did you resolved the problem?

ImJCabus, 2017-11-18:
Nope. Getting no answer at all to my question here kind of discouraged me...

sandeepmistry, 2018-01-29:
Please see https://github.com/sandeepmistry/noble/issues/523#issuecomment-309094112

closed 2018-01-29 by sandeepmistry

charlieuniform, 2018-04-16:
Not sure if you got and answer but I tried this after reading the referenced threads. I added:
```
After=hciuart.service
```
to the service file. Also I have the bluetooth service disabled per bleno recommendation.

stidiovip, 2018-10-16:
I have read all of this but I can't clearly se the working answere ?
HIs there any working solution for this ?

virtuosonic, 2023-12-29:
Hi

I'm not sure if this is the best or at least correct way to do this, but it worked for me on Raspberry pi 3B+ running Raspbian stretch, here is my service file

```
[Unit]
Description=Sexbot bluetooth LE peripheral
After=sys-subsystem-bluetooth-devices-hci0.device
Wants=sys-subsystem-bluetooth-devices-hci0.device
Requires=sys-subsystem-bluetooth-devices-hci0.device
```

To find what dependency to add I used _**systemd-analize**_ to generate an startup order graph using the following command

`systemd-analyze plot > startup_order.svg`

Then opened with `xdg-open startup_order.svg`

virtuosonic, 2023-12-29:
on raspbian-lite I had to delay more the start so the service file ended like this
```
[Unit]
Description=Sexbot bluetooth LE peripheral
After=multi-user.target
Wants=sys-subsystem-bluetooth-devices-hci0.device
Requires=bluetooth.target
```