noble/bleno #346: Invalid attribute table?

bluetooth-mdw, 2017-10-23:
I'm creating a very simple GATT peripheral which has one service with one characteristic. Android phones will not complete service discovery (using nRF Connect) and a disconnect occurs. Using a desktop tool ("btool" from TI) I can perform service discovery but it looks like there are two instances of the Generic Attribute Service, one of which is incomplete and an attempt to read my one and only characteristic generates an INVALID HANDLE error.

```
var bleno = require('bleno');

bleno.on('stateChange', function (state) {
  console.log('on -> stateChange: ' + state);

  if (state === 'poweredOn') {
    bleno.startAdvertising('BBC micro:bit [xxxxx]');
  } else {
    bleno.stopAdvertising();
  }
});

bleno.on('accept', function (clientAddress) {
  console.log('on -> accept, client: ' + clientAddress);
  bleno.updateRssi();
});

bleno.on('rssiUpdate', function (rssi) {
  console.log('on -> rssiUpdate: ' + rssi);
});

bleno.on('advertisingStart', function (error) {
  console.log('on -> advertisingStart: ' + (error ? 'error ' + error : 'success'));

  if (!error) {
    bleno.setServices([
      new bleno.PrimaryService({
        uuid: '180a',
        characteristics: [
          // Model Number String
          new bleno.Characteristic({
            value: null,
            uuid: '2a24',
            properties: ['read'],
            // Send a message back to the client with the characteristic's value
            onReadRequest: function (offset, callback) {
              console.log("Read request received");
              callback(this.RESULT_SUCCESS, new Buffer("BBC micro:bit"));
            }
          })
        ]
      })
    ]);
  }
});

bleno.on('servicesSet', function(error) {
  console.log('on -> servicesSet: ' + (error ? 'error ' + error : 'success'));
});
```

Console output:
```
on -> stateChange: poweredOn
on -> advertisingStart: success
on -> servicesSet: success
on -> accept, client: 78:c5:e5:70:7e:2c
on -> rssiUpdate: 127
Read request received
```

[image]

bluetooth-mdw, 2017-10-26:
After a reboot this issue seems to have vanished. Mysterious but I'm closing. Will reopen only if it starts to materialise again and I have more information.

bluetooth-mdw, 2017-10-26:
Reopening this. Changed name to reflect new information. 

The same or similar issue keeps occurring. I can connect to the device running my script (RaspBerry Pi) but get disconnected with service discovery having failed. It worked fine for about 30 minutes but has stopped working again. No amount of rebooting has helped and I've upgraded to the latest version of BlueZ (5.47).

My protocol analyser just shows this:

[image]

indicating the Pi is disconnecting but giving no clues as to why.

Code currently looks like this:

```
// Using the bleno module
var bleno = require('bleno');
 
// Once bleno starts, begin advertising our BLE address
bleno.on('stateChange', function(state) {
    console.log('State change: ' + state);
    if (state === 'poweredOn') {
        bleno.startAdvertising('MyDevice',['12ab']);
    } else {
        bleno.stopAdvertising();
    }
});
 
// Notify the console that we've accepted a connection
bleno.on('accept', function(clientAddress) {
    console.log("Accepted connection from address: " + clientAddress);
});
 
// Notify the console that we have disconnected from a client
bleno.on('disconnect', function(clientAddress) {
    console.log("Disconnected from address: " + clientAddress);
});
 
// When we begin advertising, create a new service and characteristic
bleno.on('advertisingStart', function(error) {
    if (error) {
        console.log("Advertising start error:" + error);
    } else {
        console.log("Advertising start success");
        bleno.setServices([
            
            // Define a new service
            new bleno.PrimaryService({
                uuid : '12ab',
                characteristics : [
                    
                    // Define a new characteristic within that service
                    new bleno.Characteristic({
                        value : null,
                        uuid : '34cd',
                        properties : ['notify', 'read', 'write'],
                        
                        // If the client subscribes, we send out a message every 1 second
                        onSubscribe : function(maxValueSize, updateValueCallback) {
                            console.log("Device subscribed");
                            this.intervalId = setInterval(function() {
                                console.log("Sending: Hi!");
                                updateValueCallback(new Buffer("Hi!"));
                            }, 1000);
                        },
                        
                        // If the client unsubscribes, we stop broadcasting the message
                        onUnsubscribe : function() {
                            console.log("Device unsubscribed");
                            clearInterval(this.intervalId);
                        },
                        
                        // Send a message back to the client with the characteristic's value
                        onReadRequest : function(offset, callback) {
                            console.log("Read request received");
                            callback(this.RESULT_SUCCESS, new Buffer("Echo: " + 
                                    (this.value ? this.value.toString("utf-8") : "")));
                        },
                        
                        // Accept a new value for the characterstic's value
                        onWriteRequest : function(data, offset, withoutResponse, callback) {
                            this.value = data;
                            console.log('Write request: value = ' + this.value.toString("utf-8"));
                            callback(this.RESULT_SUCCESS);
                        }
 
                    })
                    
                ]
            })
        ]);
    }
});
```

nRf Connect log:

[image]

bluetooth-mdw, 2017-10-26:
Seems to be working fine on my Raspberry Pi 3. The issues are occurring on a Raspberry Pi 2 with LE dongle.

sandeepmistry, 2018-01-29:
@bluetooth-mdw please see https://github.com/sandeepmistry/bleno#linux

If you've followed the instructions, and it's still not working please provide and re-open:

* Kernel version
* btmon or hcidump capture

closed 2018-01-29 by sandeepmistry