noble/bleno #348: Workaround for Pi Zero W installation

akmandala, 2017-10-30:
Hi, I've been manage to install the library on Pi Zero W, but to do so I need to do all pre-requisite (including updating node) on Pi Zero W, THEN do bleno installation on Pi 3, then bring back the sdcard to Pi Zero W. 
Anyone experiencing the same, and find a way to do full pi zero W install?

keeslinp, 2017-11-03:
I was able to just run `npm i` for the rpi0w. What is the error that you're getting which prevents you from doing so?

akmandala, 2017-11-03:
Sorry could not be more detail, but it just stuck for hours when I do sudo
npm install bleno, with or without -g., it is approximately 2/3 of progress
bar.

The environment is either latest stretch or jessie, both the same, after
doing apt-get update && upgrade, and also install latest node, then
following pre-requisite steps on this github readme.

keeslinp, 2017-11-03:
I do remember that it took a few hours when I installed it, the pi0 isn't a very powerful device. What I ended up doing was cross-compiling when I created my image using yocto. That might not fit your use case though. I'd recommend just leaving it overnight just in case it is an issue of just being super slow.

Maybe try running it with `--verbose` to see if that gives more information?

akmandala, 2017-11-03:
Ah yes, will do that and report back, need to build fresh one anyway.

akmandala, 2017-11-14:
Sorry a bit a while on this, it is definitely work on Pi Zero W, but took much longer time, hours, compare to minutes if you do it on Pi 3. Given that A correct Node & NPM already installed on Pi Zero, I guess if someone want to save a bit of time, doing all installation on Pi 3 (if they have one).

closed 2017-11-14 by akmandala