noble/bleno #354: Using "Long" Write Characteristics

vicatcu, 2017-11-27:
I know there are [two](https://github.com/sandeepmistry/bleno/issues/59) [other](https://github.com/sandeepmistry/bleno/issues/112) (closed) issues related to this, but I'm new here, and I couldn't make sense of the discussions in those issues. 

I'm using a Raspberry Pi 3, and trying to use BLE as a means to set / read some values on the Pi from my phone. Here's an example of such a value (spinning off the test.js in the repo).

```javascript
var Password_Characteristic = function() {
  Password_Characteristic.super_.call(this, {
    uuid: 'fffffffffffffffffffffffffffffff4',
    properties: ['write', 'writeWithoutResponse']
  });
};

util.inherits(Password_Characteristic, BlenoCharacteristic);

Password_Characteristic.prototype.onWriteRequest = function(data, offset, withoutResponse, callback) {
  console.log('Password_Characteristic write request: ' + data.toString() + ' ' + offset + ' ' + withoutResponse);

  callback(this.RESULT_SUCCESS);
};
```

I'm using LightBlue on Android to play with this idea. The problem that I'm experiencing is that if I send a long value (> 20 bytes) the console.log on the Raspberry pi shows a truncated write. So if I send "ABCDEFGHIJKLMNOPQRSTUVWXYZ" from LightBlue, the Raspberry Pi prints out only "ABCDEFGHIJKLMNOPQRST" (i.e. the first 20 bytes of the string). Specifically the following is printed:

`Password_Characteristic write request: ABCDEFGHIJKLMNOPQRST 0 true`

I know if I want up to 40 characters (instead of up to 20), I can make two characteristics, one for each half. But maybe there's just something I need to do to more correctly define the write-able characteristic in Bleno? Or is this actually a deficiency in LightBlue? Hopefully someone can help steer me in the right direction.

elovin, 2017-11-28:
This is also from interest to me, I will check how many characters I can write to my characteristic tomorrow and report back here.

vicatcu, 2017-12-12:
Yea I've scoured the web and come across those links as well. I have no problem with "long reads"... but I've only managed to work around the "long writes" limitation by modeling my characteristic as two characteristics if it won't fit into 20 bytes, but will fit into 40 bytes...

sandeepmistry, 2018-01-29:
Are you sure the Android side is doing a long write?

Please provide a btmon or hcidump capture and re-open.

closed 2018-01-29 by sandeepmistry