noble/bleno #356: No service discovered with nRF Connect Android App

rena2019, 2017-12-12:
I setup bleno on an Ubuntu 16.04 system and trying to discover the examples/battery-service with the nRF Connect App under Android. The device scan shows the 'Battery' device but no service can be discovered. This is the log of the nRF Connect App:

nRF Connect, 2017-12-12
Battery (XX:XX:XX:XX:XX:XX)
V	10:37:54.807	Connecting to XX:XX:XX:XX:XX:XX...
D	10:37:54.807	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE)
D	10:37:54.869	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
D	10:37:54.872	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	10:37:54.872	Connected to XX:XX:XX:XX:XX:XX
V	10:37:54.900	Discovering services...
D	10:37:54.900	gatt.discoverServices()
D	10:38:25.277	[Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
D	10:38:25.282	[Callback] Connection state changed with status: 22 and new state: DISCONNECTED (0)
I	10:38:25.284	Disconnected
E	10:38:25.292	Error 22 (0x16): GATT CONN TERMINATE LOCAL HOST

nkolban, 2017-12-25:
I see to be having a similar issue.   When I run nRFConnect I also see no services.  Here is the nRFConnect trace:

```
V	15:52:49.675	Connecting to 00:1A:7D:DA:71:0E...
D	15:52:49.675	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE)
D	15:52:50.266	[Server callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	15:52:50.266	[Server] Device with address 00:1A:7D:DA:71:0E connected
V	15:52:50.266	[Server] Creating server connection...
D	15:52:50.266	server.connect(device, autoConnect = false)
I	15:52:50.278	[Server] MTU changed to 517
D	15:52:50.366	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
D	15:52:50.370	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	15:52:50.370	Connected to 00:1A:7D:DA:71:0E
V	15:52:50.384	Discovering services...
D	15:52:50.384	gatt.discoverServices()
D	15:52:50.415	[Callback] Services discovered with status: 0
I	15:52:50.415	Services discovered
V	15:52:50.415	No services found
```

and the application I am running:

```
var bleno = require("bleno");
var name = "BLENO";
bleno.on("stateChange", (state) => {
	console.log("onStateChange: " + state);
	if (state === "poweredOn") {
		bleno.startAdvertising(name, ["91bad492b9504226aa2b4ede9fa42f59"], (error) => {
			console.log("Advertising cb: " + error);
		});
		console.log("Started advertising");
	}
});
bleno.on("advertisingStart", (error) => {
	console.log("Adverising started!: " + error);
	bleno.setServices([ new bleno.PrimaryService({
		uuid: "91bad492b9504226aa2b4ede9fa42f59",
		characteristics: [ new bleno.Characteristic({
			uuid: "4fafc2011fb5459e8fccc5c9c331914b",
			properties: ["read", "write", "writeWithoutResponse", "notify", "indicate"],
			value: null
		})]

	})], (error) => {
		console.log("setServices: " + error);
	});
});

```

acube-tech, 2018-01-14:
Hi,
Same issue here using a Raspberry PI and nRFconnect on an Android device : 

D	21:49:49.653	gatt.close()
D	21:49:49.656	wait(200)
V	21:49:49.860	Connecting to B8:27:EB:A3:27:11...
D	21:49:49.860	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE)
D	21:49:50.097	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	21:49:50.097	Connected to B8:27:EB:A3:27:11
D	21:49:50.116	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
V	21:49:50.144	Discovering services...
D	21:49:50.144	gatt.discoverServices()
D	21:50:05.260	[Callback] Services discovered with status: 129
E	21:50:05.260	Error 129 (0x81): GATT INTERNAL ERROR
D	21:50:05.268	[Callback] Connection state changed with status: 22 and new state: DISCONNECTED (0)
E	21:50:05.268	Error 22 (0x16): GATT CONN TERMINATE LOCAL HOST
I	21:50:05.268	Disconnected
D	21:50:05.319	[Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED

The whole connection freezes during "gatt.discoverServices()" for 15seconds. 
The funny part is that my whole JS script freezes during 15 seconds : I tried to set an interval `setInterval(test, 500)`
Where test is a function that logs something on the terminal. Even the setInterval stops during this 15 seconds freeze !

Secondly, I tested to connect to the raspi, using nrfconnect on three different phones: iPhone 7, Galaxy S8 and OnePlus2. 
This issue only occurs with the OnePlus 2 (Android 6.0.1) and not with the 2 other phones where the connection is OK immediately. 

So is this a BLENO or a phone issue ?

Thanks for your kind help !

acube-tech, 2018-01-15:
Hi there,

I found something during my multiple tests: 

1. If I start my script on my terminal, nrfconnect never connects to bleno
2. If I launch two terminals and **launch the same script on both terminals, it works** : nrfconnect is able to connect to the first script I launched on the first terminal !

@rena2019 Can you try the same exercice I tell us your result ? Does it work if you have your script running in the same time on two terminals ?

sandeepmistry, 2018-01-29:
Please see https://github.com/sandeepmistry/bleno#linux

If you've followed the instructions, and it's still not working please provide and re-open:

* Kernel version
* btmon or hcidump capture

closed 2018-01-29 by sandeepmistry

acube-tech, 2018-01-30:
Hi @sandeepmistry 
I cannot reopen this issue because I didn't created it. Can you do it please or should I open a new issue ?

--> I confirm I still have the same issue following the instructions you mentioned. 

Here the information you asked: 

**Kernel version**
```
pi@raspi3:~ $ uname -a
Linux raspi3 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux
```

**btmon capture**
```
Bluetooth monitor ver 5.23
* Unknown packet (code 12 len 34)                                                                    [hci65535] 0.349976
        4c 69 6e 75 78 20 76 65 72 73 69 6f 6e 20 34 2e  Linux version 4.
        39 2e 33 35 2d 76 37 2b 20 28 61 72 6d 76 37 6c  9.35-v7+ (armv7l
        29 00                                            ).
* Unknown packet (code 12 len 33)                                                                    [hci65535] 0.349985
        42 6c 75 65 74 6f 6f 74 68 20 73 75 62 73 79 73  Bluetooth subsys
        74 65 6d 20 76 65 72 73 69 6f 6e 20 32 2e 32 32  tem version 2.22
        00                                               .
= New Index: B8:27:EB:A3:27:11 (BR/EDR,UART,hci0)                                                        [hci0] 0.349989
* Unknown packet (code 8 len 0)                                                                          [hci0] 0.349992
* Unknown packet (code 10 len 8)                                                                         [hci0] 0.349994
        11 27 a3 eb 27 b8 0f 00                          .'..'...
* Unknown packet (code 14 len 30)                                                                        [hci0] 0.349998
        03 00 00 00 00 00 02 16 00 01 00 00 00 10 6e 6f  ..............no
        64 65 00 00 00 00 00 00 00 00 00 00 00 00        de............
* Unknown packet (code 14 len 30)                                                                    [hci65535] 0.350001
        02 00 00 00 02 00 01 0e 00 01 00 00 00 10 6e 6f  ..............no
        64 65 00 00 00 00 00 00 00 00 00 00 00 00        de............
* Unknown packet (code 14 len 30)                                                                    [hci65535] 0.350045
        01 00 00 00 02 00 01 0e 00 01 00 00 00 10 62 74  ..............bt
        6d 6f 6e 00 00 00 00 00 00 00 00 00 00 00        mon...........
> HCI Event: LE Meta Event (0x3e) plen 19                                                                [hci0] 8.499626
      LE Connection Complete (0x01)
        Status: Success (0x00)
        Handle: 64
        Role: Slave (0x01)
        Peer address type: Random (0x01)
        Peer address: 4E:B2:BC:C7:F2:97 (Resolvable)
        Connection interval: 45.00 msec (0x0024)
        Connection latency: 0.00 msec (0x0000)
        Supervision timeout: 20000 msec (0x07d0)
        Master clock accuracy: 0x00
* Unknown packet (code 17 len 19)                                                                        [hci0] 8.499712
        01 00 00 00 0b 00 97 f2 c7 bc b2 4e 02 00 00 00  ...........N....
        00 00 00                                         ...
* Unknown packet (code 17 len 19)                                                                        [hci0] 8.499712
        02 00 00 00 0b 00 97 f2 c7 bc b2 4e 02 00 00 00  ...........N....
        00 00 00                                         ...
< HCI Command: LE Read Remote Used Features (0x08|0x0016) plen 2                                         [hci0] 8.499992
        Handle: 64
@ Device Connected: 4E:B2:BC:C7:F2:97 (2) flags 0x0000
> HCI Event: Command Status (0x0f) plen 4                                                                [hci0] 8.501300
      LE Read Remote Used Features (0x08|0x0016) ncmd 1
        Status: Success (0x00)
* Unknown packet (code 14 len 30)                                                                    [hci65535] 8.509075
        04 00 00 00 00 00 02 16 00 01 00 00 00 10 68 63  ..............hc
        69 63 6f 6e 66 69 67 00 00 00 00 00 00 00        iconfig.......
* Unknown packet (code 15 len 4)                                                                     [hci65535] 8.509140
        04 00 00 00                                      ....
> ACL Data RX: Handle 64 flags 0x02 dlen 11                                                              [hci0] 8.686703
      ATT: Read By Group Type Request (0x10) len 6
        Handle range: 0x0001-0xffff
        Attribute group type: Primary Service (0x2800)
> ACL Data RX: Handle 64 flags 0x02 dlen 11                                                             [hci0] 13.636705
[... 35 lines cut ...]
      ATT: Read By Group Type Response (0x11) len 13
        Attribute data length: 6
        Attribute data list: 2 entries
        Handle: 0x0001
        Value: 05000018
        Handle: 0x0006
        Value: 09000118
< ACL Data TX: Handle 64 flags 0x00 dlen 18                                                             [hci0] 23.917306
      ATT: Read By Group Type Response (0x11) len 13
        Attribute data length: 6
        Attribute data list: 2 entries
        Handle: 0x0001
        Value: 05000018
        Handle: 0x0006
        Value: 09000118
< HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1                                             [hci0] 23.917322
        Advertising: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                                             [hci0] 23.921061
      LE Set Advertise Enable (0x08|0x000a) ncmd 1
        Status: Success (0x00)
> HCI Event: Number of Completed Packets (0x13) plen 5                                                  [hci0] 24.161337
        Num handles: 1
        Handle: 64
        Count: 3
```

rena2019, 2018-02-05:
Hi all,
I still have the same issue. Please see below for my btmon log. I cannot reopen this issue, too (can not find an option on this web page to re-open the issue)

$ uname -a
Linux vbox 4.10.0-42-generic #46~16.04.1-Ubuntu SMP Mon Dec 4 15:57:59 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

$ btmon --write ~/hcitrace.snoop
Bluetooth monitor ver 5.37
= Note: Linux version 4.10.0-42-generic (x86_64)                                                                                                                  0.427422
= Note: Bluetooth subsystem version 2.22                                                                                                                          0.427425
= New Index: 5C:F3:70:61:B7:73 (BR/EDR,USB,hci0)                                                                                                           [hci0] 0.427426
= Open Index: 5C:F3:70:61:B7:73                                                                                                                            [hci0] 0.427427
= Index Info: 5C:F3:70:61:B7:73 (Broadcom Corporation)                                                                                                     [hci0] 0.427427
* Unknown packet (code 14 len 30)                                                                                                                                 0.427428
        01 00 00 00 02 00 01 0e 00 01 00 00 00 10 62 6c  ..............bl
        75 65 74 6f 6f 74 68 64 00 00 00 00 00 00        uetoothd......  
* Unknown packet (code 14 len 30)                                                                                                                                 0.427438
        02 00 00 00 02 00 01 0e 00 01 00 00 00 10 62 74  ..............bt
        6d 6f 6e 00 00 00 00 00 00 00 00 00 00 00        mon...........  
* Unknown packet (code 14 len 30)                                                                                                                                16.007704
        03 00 00 00 02 00 01 0e 00 01 00 00 00 10 6e 6f  ..............no
        64 65 00 00 00 00 00 00 00 00 00 00 00 00        de............  
* Unknown packet (code 14 len 30)                                                                                                                                16.019121
        04 00 00 00 00 00 02 16 00 01 00 00 00 10 6e 6f  ..............no
        64 65 00 00 00 00 00 00 00 00 00 00 00 00        de............  
* Unknown packet (code 15 len 4)                                                                                                                                 16.019410
        04 00 00 00                                      ....            
* Unknown packet (code 14 len 30)                                                                                                                         [hci0] 16.019412
        04 00 00 00 00 00 02 16 00 01 00 00 00 10 6e 6f  ..............no
        64 65 00 00 00 00 00 00 00 00 00 00 00 00        de............  
< HCI Command: Set Event Mask (0x03|0x0001) plen 8                                                                                                        [hci0] 16.021167
        Mask: 0x3dbff807fffbffff
          Inquiry Complete
          Inquiry Result
          Connection Complete
          Connection Request
          Disconnection Complete
          Authentication Complete
          Remote Name Request Complete
          Encryption Change
          Change Connection Link Key Complete
          Master Link Key Complete
          Read Remote Supported Features Complete
          Read Remote Version Information Complete
          QoS Setup Complete
          Command Complete
          Command Status
          Hardware Error
          Flush Occurred
          Role Change
          Mode Change
          Return Link Keys
          PIN Code Request
          Link Key Request
          Link Key Notification
          Loopback Command
          Data Buffer Overflow
          Max Slots Change
          Read Clock Offset Complete
          Connection Packet Type Changed
          QoS Violation
          Page Scan Mode Change
          Page Scan Repetition Mode Change
          Flow Specification Complete
          Inquiry Result with RSSI
          Read Remote Extended Features Complete
          Synchronous Connection Complete
          Synchronous Connection Changed
          Sniff Subrating
          Extended Inquiry Result
[... 81 lines cut ...]
          Battery Service (0x180f)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                               [hci0] 16.071281
      LE Set Advertising Data (0x08|0x0008) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1                                                                                               [hci0] 16.071330
        Advertising: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                               [hci0] 16.075426
      LE Set Advertise Enable (0x08|0x000a) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Response Data (0x08|0x0009) plen 32                                                                                            [hci0] 16.075553
        Length: 9
        Name (short): Battery
[... cut ...]
< HCI Command: LE Set Advertising Data (0x08|0x0008) plen 32                                                                                              [hci0] 16.078491
        Length: 7
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        16-bit Service UUIDs (complete): 1 entry
          Battery Service (0x180f)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                               [hci0] 16.081244
      LE Set Advertising Data (0x08|0x0008) ncmd 1
        Status: Success (0x00)

sandeepmistry, 2018-03-12:
@rena2019 it still looks like `bluetoothd` is running, unless your bleno app sets the battery service.

Please share your bleno app if you have stopped `bluetoothd`