noble/bleno #359: Accepting connection after Discovery

bwenninger, 2017-12-21:
Our Android app allows the user to find Bluetooth devices so that it can communicate with our application running on a Raspberry PI.  On the PI we are supporting classic Bluetooth connections and low energy connections and are using Bleno for the BLE side.  When the user on Android searches for nearby devices our application calls StartDiscovery on the BluetoothAdapter of the Android device.  During discovery the BLE connection on the PI is found but we ignore it (we are using BLE for connecting iOS devices not Android).  However, at the end of the discovery session, as in when Android alerts our app that discovery is finished, Bleno accepts a connection to the Android device on the PI.  Below are the debug statements from Bleno/node.

Is there a way that we can configure Bleno to not do the accept?  Or is there something from the logs below that we should be looking for to know that this connection is not really a good connection?

Thanks!

Thu, 21 Dec 2017 18:59:17 GMT hci onSocketData: 043e1301004000010037af982c2f6c27000000d00705
Thu, 21 Dec 2017 18:59:17 GMT hci       event type = 4
Thu, 21 Dec 2017 18:59:17 GMT hci       sub event type = 62
Thu, 21 Dec 2017 18:59:17 GMT hci               LE meta event type = 1
Thu, 21 Dec 2017 18:59:17 GMT hci               LE meta event status = 0
Thu, 21 Dec 2017 18:59:17 GMT hci               LE meta event data = 4000010037af982c2f6c27000000d00705
Thu, 21 Dec 2017 18:59:17 GMT hci                       handle = 64
Thu, 21 Dec 2017 18:59:17 GMT hci                       role = 1
Thu, 21 Dec 2017 18:59:17 GMT hci                       address type = public
Thu, 21 Dec 2017 18:59:17 GMT hci                       address = 6c:2f:2c:98:af:37
Thu, 21 Dec 2017 18:59:17 GMT hci                       interval = 48.75
Thu, 21 Dec 2017 18:59:17 GMT hci                       latency = 0
Thu, 21 Dec 2017 18:59:17 GMT hci                       supervision timeout = 20000
Thu, 21 Dec 2017 18:59:17 GMT hci                       master clock accuracy = 5
Thu, 21 Dec 2017 18:59:17 GMT hci onSocketData: 040f0400011620
Thu, 21 Dec 2017 18:59:17 GMT hci       event type = 4
Thu, 21 Dec 2017 18:59:17 GMT hci       sub event type = 15
Thu, 21 Dec 2017 18:59:17 GMT mgmt on data ->0b0000000d0037af982c2f6c01000000000000
Thu, 21 Dec 2017 18:59:17 GMT hci onSocketData: 0240200b0007000400080100ffff002a
Thu, 21 Dec 2017 18:59:17 GMT hci       event type = 2
Thu, 21 Dec 2017 18:59:17 GMT hci               cid = 4
Thu, 21 Dec 2017 18:59:17 GMT hci               handle = 64
Thu, 21 Dec 2017 18:59:17 GMT hci               data = 080100ffff002a
Thu, 21 Dec 2017 18:59:17 GMT gatt handing request: 080100ffff002a
Thu, 21 Dec 2017 18:59:17 GMT gatt read by type: startHandle = 0x1, endHandle = 0xffff, uuid = 0x2a00
Thu, 21 Dec 2017 18:59:17 GMT gatt read by type response: 0910030031372d4b4744422d353331323334
Thu, 21 Dec 2017 18:59:17 GMT gatt send: 0910030031372d4b4744422d353331323334
Thu, 21 Dec 2017 18:59:17 GMT hci write acl data pkt - writing: 0240001600120004000910030031372d4b4744422d353331323334
Thu, 21 Dec 2017 18:59:17 GMT hci onSocketData: 043e0c040040000f00000000000000
Thu, 21 Dec 2017 18:59:17 GMT hci       event type = 4
Thu, 21 Dec 2017 18:59:17 GMT hci       sub event type = 62
Thu, 21 Dec 2017 18:59:17 GMT hci               LE meta event type = 4
Thu, 21 Dec 2017 18:59:17 GMT hci               LE meta event status = 0
Thu, 21 Dec 2017 18:59:17 GMT hci               LE meta event data = 40000f00000000000000

bwenninger, 2018-01-11:
Has anyone else run into this issue or found a solution?

sandeepmistry, 2018-01-29:
> Is there a way that we can configure Bleno to not do the accept? Or is there something from the logs below that we should be looking for to know that this connection is not really a good connection?

Nope.

closed 2018-01-29 by sandeepmistry