noble/bleno #362: XPC binding broken on MacOS 10.13 (High Sierra)

pavel-zdenek, 2018-01-11:
From what i found out quickly, BLE state change is now id 4 not 6, and when i changed just that to start advertising, there was no XPC response to message id 8. I guess this approach is not long term sustainable, which is probably why you started effort branch `core-bluetooth`. I have seen your comment somewhere, that you plan to merge the branch, but it is more than 2yrs old now. I have a mild interest in getting it working on MacOS, so can you tell me what is the state of the branch and its mergeability by an outsider?

ferencbrachmann, 2018-01-12:
Can you comment on High Sierra support in general? Thousands of people using https://github.com/dermike/slide-beacon-app/ is waiting on you to add High Sierra support :)

jacobrosenthal, 2018-01-12:
Please use existing issue #341, (tldr test #360)

And as always, show support to dependencies you rely upon, were up on opencollective at https://opencollective.com/noble

closed 2018-01-12 by jacobrosenthal