noble/bleno #363: No compatible USB Bluetooth 4.0 device in Virtual Machine.

jsho5150, 2018-01-29:
I have a bleno device working on a Windows 7 PC without issue, but when I try to configure/run that same device from within a Virtual Machine, I get the "No compatible device" message.  I was able to install the required software, including WinUSB in the VM and can see the device (and its ID's) inside the VM as well, but can't seem to get bleno (bluetooth_hci_socket) to find the device no matter what I try.  Has anyone had any luck running bleno from inside a VM in the past that could provide some tips?  (Or is this just not possible?)

Thanks,
Jason

jsho5150, 2018-01-29:
Some additional information:

By debugging my bleno project, I was able to determine that only a single USB device is being returned by the usb.getDeviceList function in usb.js.  Based on the vendor id of the device (0x1D6B) it appears to be identified as a USB hub.  After some additional investigation, I came across the following update to libusb which may resolve my issue: https://github.com/libusb/libusb/pull/257/commits/dca85e3c39c558f628b4641de64617a9fac1bbb3

That change seems to ignore the lack of a port number (SPDRP_ADDRESS) for a returned device.  I'm pretty sure this would move me a step closer to resolving my issue, but I'm not 100% sure how to get this libusb fix into my bleno project.  (I've looked at the libusb code on my system, and it does NOT contain this fix.) 

My other short term approach will be to see if I can force a USB hub to have a port number in Windows.  (Doubtful I'll be able to do that, so any help on getting libusb updated for my bleno based program would be appreciated.

sandeepmistry, 2018-01-31:
You'll need to get the maintainers of node-usb to get this change in their module.

https://github.com/tessel/node-usb

closed 2018-01-31 by sandeepmistry