noble/bleno #365: BLE not recognized on Atheros AR6233 (Digi Connect Core 6 SBC)

h0ru5, 2018-01-31:
it seems bleno does not recognize the BLE capability of this chipset.
running echo example gives this error:
```
root@ccimx6sbc:/usr/lib/node_modules/bleno/examples/echo# BLENO_HCI_DEVICE_ID=0 node main.js 
bleno - echo
bleno warning: adapter does not support Bluetooth Low Energy (BLE, Bluetooth Smart).
               Try to run with environment variable:
               [sudo] BLENO_HCI_DEVICE_ID=x node ...
on -> stateChange: unsupported
```
However, the chipset does support BLE. E.g. `hcitool lescan` works.

The output of `hciconfig hci0 lestates` is this:
```
Supported link layer states:
	YES Non-connectable Advertising State
	YES Scannable Advertising State
	YES Connectable Advertising State
	YES Directed Advertising State
	YES Passive Scanning State
	YES Active Scanning State
	YES Initiating State/Connection State in Master Role
	YES Connection State in the Slave Role
	YES Non-connectable Advertising State and Passive Scanning State combination
	YES Scannable Advertising State and Passive Scanning State combination
	YES Connectable Advertising State and Passive Scanning State combination
	YES Directed Advertising State and Passive Scanning State combination
	NO  Non-connectable Advertising State and Active Scanning State combination
	NO  Scannable Advertising State and Active Scanning State combination
	NO  Connectable Advertising State and Active Scanning State combination
	NO  Directed Advertising State and Active Scanning State combination
	NO  Non-connectable Advertising State and Initiating State combination
	NO  Scannable Advertising State and Initiating State combination
	NO  Non-connectable Advertising State and Master Role combination
	NO  Scannable Advertising State and Master Role combination
	NO  Non-connectable Advertising State and Slave Role combination
	NO  Scannable Advertising State and Slave Role combination
	NO  Passive Scanning State and Initiating State combination
	NO  Active Scanning State and Initiating State combination
	NO  Passive Scanning State and Master Role combination
	NO  Active Scanning State and Master Role combination
	NO  Passive Scanning State and Slave Role combination
	NO  Active Scanning State and Slave Role combination
	NO  Initiating State and Master Role combination/Master Role and Master Role combination
```

h0ru5, 2018-02-01:
it seems that `bluetooh-hci-socket` works, I can run the advertisement sample without problems.

https://github.com/sandeepmistry/node-bluetooth-hci-socket/blob/master/examples/le-advertisement-test.js

sandeepmistry, 2018-02-05:
Could you please provide debug logs?

Maybe this check doesn't match? https://github.com/sandeepmistry/bleno/blob/master/lib/hci-socket/hci.js#L645

h0ru5, 2018-02-05:
> Maybe this check doesn't match? https://github.com/sandeepmistry/bleno/blob/master/lib/hci-socket/hci.js#L645

exactly. it works fine if I "patch" this check to  `if (hciVer < 0x05) {`

It's not easy to provide the debug logs unfortunately, as the yocto-built node.js  is compiled without debugging support

sandeepmistry, 2018-03-12:
@h0ru5 

> It's not easy to provide the debug logs unfortunately, as the yocto-built node.js is compiled without debugging support

You need you just run with `sudo DEBUG=* node <file>.js`