noble/bleno #366: How can I change the TX address

bluetiger30, 2018-02-14:
Hi, 
how can I change the TX address? 
I noticed that there is address change event in the bleno file which listens to the address of the BLE device, is there a way to possibly be able to change that address?

I'm using the following BLE module if that helps :
https://www.asus.com/uk/Networking/USBBT400/
if its easier to change the TX address of the BLE module on a raspberry pi 3 then I would love to do that?

nemr-maher, 2018-02-20:
alternativly , when I set the foloowing in the hci.js file :
`cmd.writeUInt8(0x01, 9); // own addr typ
`
it changed the TX address to 00000000 , can I change that to something either than 00000000?

bluetiger30, 2018-02-20:
alternativly , when I set the foloowing in the hci.js file :
cmd.writeUInt8(0x01, 9); // own addr typ
it changed the TX address to 00000000 , can I change that to something other than 00000000?

sandeepmistry, 2018-02-26:
This is not supported by this library.

closed 2018-02-26 by sandeepmistry