noble/bleno #374: Device Name (OS X)

leonardofreitass, 2018-04-16:
The `BLENO_DEVICE_NAME` is not working on OS X.
I tried both solutions:
Setting the env on runtime using
```
process.env['BLENO_DEVICE_NAME'] = name;
```

And tried also to set the env var with the startup, using `BLENO_DEVICE_NAME=name npm start`.

Both work on Linux, using nRF Connect to debug, but both shows the hostname on the device name, using a MacBook Pro as host, and the LightBlue to debug.

Uniphix, 2018-04-20:
I have had issues too with the device name being completely blank when scanning them.  Once I connect to the device and what not it will work again.

vpfaulkner, 2018-04-24:
Also experiencing this.

sandeepmistry, 2018-05-05:
This is only supported on Linux.

closed 2018-05-05 by sandeepmistry