noble/bleno #376: BLE HID Keyboard with Bleno

alanw9876, 2018-04-23:
I have been attempting to write (started anyhow) a BLE HID Keyboard using Bleno.  I think I have stuff set ok but the problem I seem to have is when I connect to IOS or Android for example, I can get Bleno to connect seemingly fine, but no key events seem to transmit.  I have setup the necessary HID service, Battery Service, and Device Information Service required for a HID Keyboard.  In addition I have setup a GAP Service as well to try to coax Bleno in telling IOS or Android this is a keyboard and use the proper icon/appearance - that is not working either but not the biggest problem.  
The real problem is, since the keyboard is a peripheral  and IOS or Android device (phone/tablet/tv, whatever) is the central, how does the notification get enabled by the central.  I know normally the client (central) would subscribe but since this is HID over GATT is there something else that sets that up?  There isn't anything to say "setup subscription for this keyboard" in settings or anything nutty like that.   In short, how do I emit keypress events to the client if the client is not subscribed?  (yeah, I know, that sounds stupid, but seriously - this is convoluted)  help! please

alanw9876, 2018-04-30:
So to update, I can connect to the "bleno emulated keyboard device".  I have defined the following routines:
hid_keyboard_in.onSubscribe = function(maxValueSize, updateValueCallback) {
    console.log('hid_keyboard_in->subscribed')
	this._updateValueCallback = updateValueCallback;
}

hid_keyboard_in.onUnsubscribe = function () {
    console.log('hid_keyboard_in->unsubscribed')
	this._updateValueCallback = null;
}

hid_keyboard_in.update = function(value) {
	this._value = value;
	console.log('hid_keyboard_in->about to send', this._value)
	if (this._updateValueCallback) {
        console.log('hid_keyboard_in->sending', this._value)
        this._updateValueCallback(this._value);
	}
}
The subscribe will never get invoked when connected to it.  Shouldn't that be happening?  btw - I have tried using iPhone, iPad and multiple Android devices.  They will connect, but that is all that occurs.  Any advice is much appreciated.

alanw9876, 2018-04-30:
Also, what is the best practice for the device (bleno) to send the key events?  Do I need to merely set the value and bleno will take care of sending through the subscription or do I need to manually invoke the callback that is set in the subscribe event?

sandeepmistry, 2018-05-05:
@alanw9876 I suggest you post on stack overflow. 

You might find some relevant code on HID over GAT here, that should be possible to convert to run on bleno: https://github.com/sandeepmistry/arduino-BLEPeripheral/tree/master/src

closed 2018-05-05 by sandeepmistry