noble/bleno #381: how to run multiple instance?

slixin, 2018-04-30:
I have run bleno sucessfully on MacOS, but Id like to know how to run multi instances on the same server? Does it support?

sandeepmistry, 2018-05-05:
@slixin have you tried just running multiple node apps? Let us know if this doesn't work.

closed 2018-05-05 by sandeepmistry

leonardofreitass, 2018-05-06:
Even running multiple node apps, it does not work. If I run a second node app, it will override the advertisement of the service in the first node app. Is this a bug or bleno really doesn't support this yet?

petersaints, 2020-07-20:
I'm currently using this fork of bleno because the "official" version does not support the latest versions of Node: https://github.com/abandonware/bleno

However, I believe that this issue applies to both versions. I'm using the "startAdvertisingIBeacon" to simulate an iBeacon, but I'm only able to run one instance of my application that does it. As soon as I run another instance the previous one stops advertising.

Is this something that can be done at the Node.js/bleno level. Or is it some inherent limitation of BlueZ or perhaps even an hardware limitation? For instance, on certain Android phones I'm able to advertise multiple iBeacon signals concurrently on an app such as this one: https://play.google.com/store/apps/details?id=net.alea.beaconsimulator&hl=en