noble/bleno #382: It stopped advertising shortly after startAdvertising().

vicary, 2018-05-02:
I am testing Bleno with Noble in a Raspberry Pi.

After I called `bleno.startAdvertising(...);` I can scan the device with an iPhone and an Android, but when I refresh the device list after initial discovery the Pi never appeared again until I reboot.

I would really like to narrow down the case, where shall I start?

sandeepmistry, 2018-05-05:
@vicary I suggest you look and the debug logs`sudo DEBUG=* node <file>.js` and hcidump/btmon captures ...

closed 2018-05-05 by sandeepmistry

vicary, 2018-05-11:
@sandeepmistry Thanks for the reply.

I added the `DEBUG=*` env var, could use some help in making sense out of it.

At the very least, assuming the default advertisement interval is 100ms, shall I be seeing logs starting with the following lines every 100ms?

```
hci set advertise enable - writing: 010a200100 +6ms
hci set advertisement parameters - writing: 0106200fa000a0000000000000000000000700 +5ms
```

If that's the case, the log is not generating at that interval. It only happened 4 times in 5 mins.