noble/bleno #383: Example echo disconnects with error 22

Domvel, 2018-05-08:
The example "Echo" on Raspberry PI with bleno 0.5.0 disconnects after about 30 seconds with `Error 22 GATT CONN TERMINATE LOCAL HOST`. And not services / characteristics are visible (discovered). 
Is this a known issue?

don, 2018-05-26:
I'd try disabling `bluetoothd`, it can cause problems with services not showing up. 

    sudo service bluetooth stop

Then manually start the bluetooth

    sudo hciconfig hci0 up

If the disconnect keeps happening, run btmon to get detailed log of the communication

    sudo btmon

If you're still not seeing all the services, apps like nRF Connect have an "Refresh Services" option which should overwrite any cached data.

I was not able to duplicate this on a Raspberry Pi 3 B+. An iPhone 7 (11.3.1) and Nexus 5X (8.1.0) stayed connected.

don, 2018-06-08:
@Domvel is this still an issue?

Domvel, 2018-06-08:
Not more followed.

closed 2018-06-08 by Domvel