noble/bleno #386: Bleno Installation Error

mblue002, 2018-05-17:
Hello,
I am very interested in running bleno on a RPI 3 B.
However, my installation of bleno is failing to build BluetoothHciSocket.o
I searched through open and closed issues and did not see this particular problem. I did run "sudo apt-get upgrade" as suggested in issue #33f.
Thanks in advance.
Mike

pi@raspberrypi:~ $ npm install bleno

> usb@1.3.1 install /home/pi/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/tessel/node-usb/releases/download/1.3.1/usb_bindings-v1.3.1-node-v64-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for usb@1.3.1 and node@10.1.0 (node-v64 ABI, glibc) (falling back to source compile with node-gyp) 
make: Entering directory '/home/pi/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
...
... Skipping 
...
make: Entering directory '/home/pi/node_modules/bluetooth-hci-socket/build'
  CXX(target) Release/obj.target/binding/src/BluetoothHciSocket.o
../src/BluetoothHciSocket.cpp: In member function ‘void BluetoothHciSocket::poll()’:
../src/BluetoothHciSocket.cpp:251:95: warning: ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)’ is deprecated (declared at ../../nan/nan.h:950) [-Wdeprecated-declarations]
     Nan::MakeCallback(Nan::New<Object>(this->This), Nan::New("emit").ToLocalChecked(), 2, argv);
                                                                                               ^
../src/BluetoothHciSocket.cpp: In member function ‘void BluetoothHciSocket::emitErrnoError()’:
../src/BluetoothHciSocket.cpp:275:72: error: no matching function for call to ‘v8::Function::NewInstance(int, v8::Local<v8::Value> [1])’
   Local<Value> error = errorConstructor->NewInstance(1, constructorArgs);
                                                                        ^
../src/BluetoothHciSocket.cpp:275:72: note: candidates are:
In file included from /home/pi/.node-gyp/10.1.0/include/node/node.h:63:0,
                 from /home/pi/.node-gyp/10.1.0/include/node/node_buffer.h:25,
                 from ../src/BluetoothHciSocket.cpp:7:
/home/pi/.node-gyp/10.1.0/include/node/v8.h:3848:44: note: v8::MaybeLocal<v8::Object> v8::Function::NewInstance(v8::Local<v8::Context>, int, v8::Local<v8::Value>*) const
   V8_WARN_UNUSED_RESULT MaybeLocal<Object> NewInstance(
                                            ^
/home/pi/.node-gyp/10.1.0/include/node/v8.h:3848:44: note:   candidate expects 3 arguments, 2 provided
/home/pi/.node-gyp/10.1.0/include/node/v8.h:3851:44: note: v8::MaybeLocal<v8::Object> v8::Function::NewInstance(v8::Local<v8::Context>) const
   V8_WARN_UNUSED_RESULT MaybeLocal<Object> NewInstance(
                                            ^
/home/pi/.node-gyp/10.1.0/include/node/v8.h:3851:44: note:   candidate expects 1 argument, 2 provided
../src/BluetoothHciSocket.cpp:282:93: warning: ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)’ is deprecated (declared at ../../nan/nan.h:950) [-Wdeprecated-declarations]
   Nan::MakeCallback(Nan::New<Object>(this->This), Nan::New("emit").ToLocalChecked(), 2, argv);
                                                                                             ^
binding.target.mk:93: recipe for target 'Release/obj.target/binding/src/BluetoothHciSocket.o' failed
make: *** [Release/obj.target/binding/src/BluetoothHciSocket.o] Error 1
make: Leaving directory '/home/pi/node_modules/bluetooth-hci-socket/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:235:12)
gyp ERR! System Linux 4.9.64-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/node_modules/bluetooth-hci-socket
gyp ERR! node -v v10.1.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/bluetooth-hci-socket):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

+ bleno@0.5.0
updated 1 package in 62.733s

georgeredinger, 2018-05-17:
> [-Wdeprecated-declarations] Nan::MakeCallback(Nan::New(this->This),
> to ‘v8::Function::NewInstance(int, v8::Localv8::Value [1])’ Local error =
> const V8_WARN_UNUSED_RESULT MaybeLocal NewInstance( ^
> Nan::MakeCallback(Nan::New(this->This), Nan::New("emit").ToLocalChecked(),

Mike,
Not completely lucid on this topic at the moment.
 There is new stuff here now. New pi zero w and pi3 + models have better
hard ware and software supourt. Posibly an out of the box solution for
those.send me some application details and I'll check what the latest stuff
will do out of the box.
George

mblue002, 2018-05-18:
Hi George,
Thanks for your reply, but I am currently interested in fixing the build failure.
Also, for clarity  I followed the instructions in the Readme.md for Raspbian:

## Ubuntu/Debian/Raspbian
### sudo apt-get install bluetooth bluez libbluetooth-dev libudev-dev
### Make sure node is on your path, if it's not, some options:

### symlink nodejs to node: sudo ln -s /usr/bin/nodejs /usr/bin/node
### install Node.js using the NodeSource package

I installed Node.js 10

Thanks,
Mike

mblue002, 2018-05-18:
Reverted to node.js 8 and now the build succeeds.
Found issued related to Node.js 10 in noble/bluetooth-hci-socket 
Closing this issue.

closed 2018-05-18 by mblue002