noble/bleno #388: Bleno installation / runtime problem

johnrmenard, 2018-05-18:
Hi,

after many attempts at installing and running Bleno on Raspberry Pi , I get this error when running projects that have run previously:

pi@raspberrypi:~/Development/BleServer $ sudo node index.js
module.js:681
  return process.dlopen(module, path._makeLong(filename));
                 ^

Error: /home/pi/Development/BleServer/node_modules/bluetooth-hci-socket/build/Release/binding.node: wrong ELF class: ELFCLASS64
    at Object.Module._extensions..node (module.js:681:18)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/home/pi/Development/BleServer/node_modules/bluetooth-hci-socket/lib/native.js:3:15)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
pi@raspberrypi:~/Development/BleServer $

johnrmenard, 2018-05-18:
This is how I installed Bleno:

sudo npm install bleno 
npm WARN rpi3-application@1.0.0 No description
npm WARN rpi3-application@1.0.0 No repository field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: mkdirp@0.5.1 (node_modules/mkdirp):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Refusing to delete /home/pi/Development/BleServer/node_modules/usb/node_modules/node-pre-gyp/node_modules/.bin/mkdirp: is outside /home/pi/Development/BleServer/node_modules/usb/node_modules/node-pre-gyp/node_modules/mkdirp and not a link
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: File exists: /home/pi/Development/BleServer/node_modules/usb/node_modules/node-pre-gyp/node_modules/.bin/mkdirp
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

+ bleno@0.3.5
removed 8 packages and updated 1 package in 16.013s
[!] 9 vulnerabilities found [285 packages audited]
    Severity: 9 Moderate
    Run `npm audit` for more detail

don, 2018-05-23:
Make sure your RPi is up to date and has the all the dependencies installed. Make sure you're running a newer version of node.

Here's how I set up bleno on a new RPi

    $ sudo apt update
    $ sudo apt dist-upgrade -y

Dev tools

    $ sudo apt install python-dev build-essential curl git mc -y

Bluetooth Dependencies
    
    $ sudo apt install bluetooth bluez libbluetooth-dev libudev-dev -y

Install Nodejs (Raspberry Pi3)

    $ curl -sL https://deb.nodesource.com/setup_8.x | sudo -E bash - 
    $ sudo apt install nodejs -y

The Raspberry Pi Zero W is ARM v6 and not supported by the Node.js package manager.

Download archive

    $ cd ~
    $ curl -O https://nodejs.org/dist/v8.9.4/node-v8.9.4-linux-armv6l.tar.xz

Install in /usr/local

    $ cd /usr/local
    $ sudo tar xf ~/node-v8.9.4-linux-armv6l.tar.xz

Create symlinks

    $ cd /usr/local/bin
    $ sudo ln -s ../node-v8.9.4-linux-armv6l/bin/node
    $ sudo ln -s ../node-v8.9.4-linux-armv6l/bin/npm

johnrmenard, 2018-05-24:
Many thanks I will try this.

johnrmenard, 2018-05-25:
That worked!

closed 2018-05-25 by johnrmenard