noble/bleno #389: raspberrypi 3 B install error

ssaylanc, 2018-05-25:
I am on the lastest raspbian strect. node version 10.2.1

`pi@raspberrypi:~ $ npm install bleno

> usb@1.3.1 install /home/pi/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/tessel/node-usb/releases/download/1.3.1/usb_bindings-v1.3.1-node-v64-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for usb@1.3.1 and node@10.2.1 (node-v64 ABI, glibc) (falling back to source compile with node-gyp)
node-pre-gyp ERR! Tried to download(undefined): https://github.com/tessel/node-usb/releases/download/1.3.1/usb_bindings-v1.3.1-node-v64-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for usb@1.3.1 and node@10.2.1 (node-v64 ABI, glibc) (falling back to source compile with node-gyp)
make: Entering directory '/home/pi/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
make: Entering directory '/home/pi/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
  AR(target) Release/obj.target/usb.a
  COPY Release/usb.a
  CXX(target) Release/obj.target/usb_bindings/src/node_usb.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
  AR(target) Release/obj.target/usb.a
  COPY Release/usb.a
  CXX(target) Release/obj.target/usb_bindings/src/node_usb.o
../src/node_usb.cc: In function 'void handleHotplug(std::pair<libusb_device*, libusb_hotplug_event>)':
../src/node_usb.cc:150:58: warning: 'v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, const char*, int, v8::Local<v8::Value>*)' is deprecated [-Wdeprecated-declarations]
  Nan::MakeCallback(Nan::New(hotplugThis), "emit", 2, argv);
                                                          ^
In file included from ../src/helpers.h:3:0,
                 from ../src/node_usb.h:21,
                 from ../src/node_usb.cc:1:
../../nan/nan.h:971:46: note: declared here
   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
                                              ^~~~~~~~~~~~
../src/node_usb.cc: In function 'void handleHotplug(std::pair<libusb_device*, libusb_hotplug_event>)':
../src/node_usb.cc:150:58: warning: 'v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, const char*, int, v8::Local<v8::Value>*)' is deprecated [-Wdeprecated-declarations]
  Nan::MakeCallback(Nan::New(hotplugThis), "emit", 2, argv);
                                                          ^
In file included from ../src/helpers.h:3:0,
                 from ../src/node_usb.h:21,
                 from ../src/node_usb.cc:1:
../../nan/nan.h:971:46: note: declared here
   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
                                              ^~~~~~~~~~~~
  CXX(target) Release/obj.target/usb_bindings/src/device.o
  CXX(target) Release/obj.target/usb_bindings/src/device.o
../src/device.cc: In static member function 'static void Req::default_after(uv_work_t*)':
../src/device.cc:237:64: warning: 'v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)' is deprecated [-Wdeprecated-declarations]
    Nan::MakeCallback(device, Nan::New(baton->callback), 1, argv);
                                                                ^
In file included from ../src/helpers.h:3:0,
                 from ../src/node_usb.h:21,
                 from ../src/device.cc:1:
../../nan/nan.h:929:46: note: declared here
   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
[... 82 lines cut ...]
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:237:12)
gyp ERR! System Linux 4.14.34-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/node_modules/bluetooth-hci-socket
gyp ERR! node -v v10.2.1
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/bluetooth-hci-socket):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

+ bleno@0.5.0
added 4 packages in 80.073s`

don, 2018-05-25:
I'm seeing those same error building dependencies with Node 10.x. I suggest you downgrade to Node 8.x. Follow these instructions https://github.com/noble/bleno/issues/388#issuecomment-391457444

rzr, 2018-06-27:
FYI, It builds fine with v8.11.3

rzr, 2018-10-24:
Relate to: https://github.com/noble/node-bluetooth-hci-socket/issues/84

akycdi, 2022-12-07:
Hi, Have you succeeded in config of pi for the use of bleno?

rzr, 2023-06-30:
May I suggest to try @abandonware fork and report at:
Relate-to: https://github.com/abandonware/bleno/issues/17