noble/bleno #390: iOS does not see services

wescur88, 2018-06-04:
I am running the battery-service example on a Linux machine running Fedora. When connecting with an iPhone it is cumbersome to connect and when you are connected, the services do not display, as they do on Android.

don, 2018-06-08:
What do you mean by "cumbersome to connect"?
Are you seeing the battery service on the iPhone?
Can you provide screenshots of Android and iOS so we can see the difference?

wescur88, 2018-06-08:
Cumbersome by it takes multiple times to connect on iOS. iPhone can only see advertised data, while Android can see advertised data as well as the services with their characteristics and descriptors. The hardware I am running on is the Samsung Artik 710.

Wesley Currie