noble/bleno #393: startAdvertising and startAdvertisingIBeacon, Command Dissallowed

lastmjs, 2018-06-08:
Hi there,

I am advertising both iBeacons and plain advertisements. I have an interval of 100 milliseconds setup, where I switch between each type of advertisement. It works very well most of the time. Every once in a while, I will get one of the following errors when I try to start advertising one of the advertisement types:

```
bleno.startAdvertisingIBeacon Error: Command Disallowed
    at Gap.onHciLeAdvertiseEnableSet (/home/pi/development/portal-blue/door-device/node_modules/bleno/lib/hci-socket/gap.js:201:15)
    at emitOne (events.js:115:13)
    at Hci.emit (events.js:210:7)
    at Hci.processCmdCompleteEvent (/home/pi/development/portal-blue/door-device/node_modules/bleno/lib/hci-socket/hci.js:669:10)
    at Hci.onSocketData (/home/pi/development/portal-blue/door-device/node_modules/bleno/lib/hci-socket/hci.js:540:12)
    at emitOne (events.js:115:13)
    at BluetoothHciSocket.emit (events.js:210:7)
```

or 

```
bleno.startAdvertising Error: Command Disallowed
    at Gap.onHciLeAdvertiseEnableSet (/home/pi/development/portal-blue/door-device/node_modules/bleno/lib/hci-socket/gap.js:201:15)
    at emitOne (events.js:115:13)
    at Hci.emit (events.js:210:7)
    at Hci.processCmdCompleteEvent (/home/pi/development/portal-blue/door-device/node_modules/bleno/lib/hci-socket/hci.js:669:10)
    at Hci.onSocketData (/home/pi/development/portal-blue/door-device/node_modules/bleno/lib/hci-socket/hci.js:540:12)
    at emitOne (events.js:115:13)
    at BluetoothHciSocket.emit (events.js:210:7)
```

I understand that you cannot advertise both at the same time, which is why I always call `bleno.stopAdvertising` before starting to advertise the other advertisement. I start up the advertising in the callback of the `bleno.stopAdvertising` call, which I would think waits until the system is ready for any other type of advertisement to begin, but that doesn't seem to be the case.

icarus31, 2022-05-05:
@lastmjs , i know it's been a while, but do you remember if you figure out something?

rzr, 2023-06-30:
May I suggest to try @abandonware fork and report at:
Relate-to: https://github.com/abandonware/bleno/issues/17