noble/bleno #397: Receive commands on one characteristic, send responses on another characteristic

EricGCooper, 2018-06-28:
Hello, I am working on an application which accepts commands on one characteristic and sends responses on a different characteristic.

The command characteristic is set to ‘write’ and onWriteRequest it parses the data and constructs a response, and then calls a sendData function exposed by the response characteristic.

The response characteristic is set to ‘read’ and ‘notify’ and in the onSubscribe function it saves the updateValueCallback in a member variable. It also implements the sendData function which takes the data as a parameter and calls the member variable update value callback function with the data. There are no errors on sending but the data does not arrive as viewed from HCI logs on an Android device.

Note that the main program creates a response characteristic and it is a parameter to the BlenoPrimaryService constructor and it is a global accessed by the command characteristic code.

I hope this makes sense and please let me know if there is more information I can provide,

Eric

EricGCooper, 2018-06-30:
I have a different take on it, I now believe it is a general notify issue.

I changed the notify so that it is sent periodically per the answer here https://stackoverflow.com/questions/28958289/how-to-implement-update-value-callback-when-peripheral-itself-is-changing-the-ch but it is still not received so it is not an issue of using different characteristic to notify. I also added a log onNotify and it is called on each notification.

I am using OSX as the peripheral and Android as the central.

Thanks again,
Eric

bbnna, 2018-11-21:
Hi, I'm working on a project and came up with the same issue. Did you find a solution in the meanwhile? 
Annika

good-lly, 2023-11-07:
@EricGCooper , @bbnna is it still an issue? So far macos 13.5 works for me. Probably - time to close this issue