noble/bleno #404: stuck on startAdvertising, MacOS 10.14

galenzhao, 2018-08-02:
bleno.on('stateChange', function(state) {
      console.log('on -> stateChange: ' + state);
    
      if (state === 'poweredOn') {
    	  console.log('start advertising');
        bleno.startAdvertising('echo', ['ec00']);
      } else {
        bleno.stopAdvertising();
      }
    });

`

    GalenZhaos-MacBook-Air:echo galenzhao$ sudo BLENO_DEVICE_NAME="custom device name" 
    BLENO_ADVERTISING_INTERVAL=500 node main.js 
    bleno - echo
    on -> stateChange: poweredOn
    start advertising

I added start event callback without called,

    bleno.on('advertisingStartError', function(error){
        console.log('on -> advertisingStart: ' + (error ? 'error ' + error : 'success'));
    	
    });
    
    bleno.on('advertisingStart', function(error) {
      console.log('on -> advertisingStart: ' + (error ? 'error ' + error : 'success'));

gregoryyoung2, 2018-08-03:
I'm getting the same thing on macOS 10.14 (18A347e), works fine on other os

chrisbartley, 2018-09-28:
noble doesn't work yet on 10.14 either, I would guess for similar reasons: https://github.com/noble/noble/pull/804

notjosh, 2018-10-02:
I think now's a good time to adopt a native approach, so I've forked https://github.com/noble/noble/issues/828 and have made some progress on porting that for Bleno.

I've spent a couple of hours on it so far, and it can establish connections + perform static reads just fine. The "read request" and "write request" events are gonna take some more noodling around (callback inception!), and I haven't looked at any of the more advanced events in of characteristics yet.

I should hopefully have a rough version working tomorrow though! At the very least, I'll clean up what I've got and push it.

🤞

notjosh, 2018-10-03:
Alright, I've got something _working_, though not _complete_ over at https://github.com/notjosh/bleno-mac. (It won't work out of the box until #415 is merged, but you can run that locally and `yarn link` it in for the meantime.)

For now, I can read/write/subscribe just fine, and it seems to be fairly stable. I haven't touched any of the iBeacon code, and some events aren't being sent.

My 90% use case works though, so hopefully it should be usable enough to start with if anyone is interested? Any help is welcome :)

notjosh, 2018-10-03:
(To be very clear: the approach was copied straight over from https://github.com/Timeular/noble-mac, so @geovie should get all the credit for the idea! I've just been filling in the gaps on the Bleno side.)

jwheatp, 2018-10-07:
@notjosh Thanks for your work ! But I'm still having the issue described by @galenzhao. Could you explain a bit more how you get the example given by galenzhao solved and working with your https://github.com/Timeular/noble-mac ? Thanks in advance !

notjosh, 2018-10-07:
@jwheatp you'll want to point to https://github.com/notjosh/bleno-mac (not the `noble-mac` that you linked to), and you'll need to either point your `bleno` instance to my branch, or manually apply the patch in https://github.com/noble/bleno/pull/415. 

Might be easier to hold off until #415 is in, 'cos that's kind of a pain to deal with just now.

karlrl, 2018-10-12:
@notjosh's `bleno-mac` is partially working for me on 10.14: advertising works, I can connect to the "device" and can read/monitor a custom characteristic using [LightBlue](https://itunes.apple.com/us/app/lightblue-explorer/id557428110) on iOS (the organization of the services is a bit wonky, though that's likely poor configuration on my part). @jwheatp how did you try to get things working? I did the following: 

- cloned https://github.com/notjosh/bleno/ and checked out the `inject-bindings` branch 
- cloned https://github.com/notjosh/bleno-mac
- modified the `package.json` of `bleno-mac` so that it pointed to my local, patched copy of `bleno` (i.e. `"dependencies": { "bleno": "/path/to/cloned/patched/bleno", ... }`)
- modified my own app's `package.json` to point to the local copy of `bleno-mac` (as in last step).

doggkruse, 2018-10-22:
One limitation of this approach using CoreBluetooth appears to be that bleno is unable to set the primary service UUID in the advertising data.  Is this expected?

notjosh, 2018-10-22:
> One limitation of this approach using CoreBluetooth appears to be that bleno is unable to set the primary service UUID in the advertising data. Is this expected?

Oh uhhh, I think that's a bug. I'll see what I can do about it, thanks!

(I've had a pause on the project that required Bleno, but I'm picking up again ~next weeks or so, so hopefully have some more updates then)

mikk92, 2018-11-14:
I just tried to install blend on 10.14. Since advertising didn't work, I found this thread.

I tried the approach from karlrl:

> - cloned https://github.com/notjosh/bleno/ and checked out the inject-bindings branch
> - cloned https://github.com/notjosh/bleno-mac
> - modified the package.json of bleno-mac so that it pointed to my [...]

Then I started `npm install bleno` in my working directory.
The command line gets stuck on `⸨░░░░░░░░░░░░░░░░░░⸩ ⠋ fetchMetadata: sill resolveWithNewModule bleno-mac@0.0.1 checking insta`

Any clues or updates? Thanks in advance

karlrl, 2018-11-15:
@mikk92 installing still works for me (though I didn't clone anything again, since they're on the same revision). What if you just try `npm install`? Does your `package.json` point to the right places?

codeofdiego, 2018-12-19:
@notjosh Thank you so much for that fix, i hope it get's merged in the original repo soon! For now, replacing the dependencies to point to your fixes made it work "out of the box".
```
"dependencies": {
    "bleno": "github:notjosh/bleno#inject-bindings",
    "bleno-mac": "github:notjosh/bleno-mac"
}
```
🚀

mark-rodgers, 2018-12-28:
@jwheatp @notjosh @karlrl Another way I got this to work in my project without having to modify my dependancy's dependancies

1. Remove both packages 

```
npm uninstall bleno bleno-mac
npm prune
```

2. Install bleno and bleno-mac from github repo, as opposed to the npm registry

```
npm i notjosh/bleno#inject-bindings notjosh/bleno-mac
```

gabceb, 2019-02-11:
I'm new to bleno running on 10.14.2 and cannot get a callback from `startAdvertising` to be executed. I've tried https://github.com/noble/bleno/issues/404#issuecomment-448626880 to get the bleno-mac deps and bleno branch. Anything im missing?

chrisco314, 2019-02-17:
Forgive the newb question @codeofdiego (I am new to node), but to run something like the echo sample (`examples/echo`), you would first create a new package file?  I see a package file at the main file, but that seems to be for the bleno library itself.  I also see one in the battery-service example, so I am assuming that would be a good starting point to clone for the other samples?  In your post, was that for your own package / project / app?

Thanks!

> @notjosh Thank you so much for that fix, i hope it get's merged in the original repo soon! For now, replacing the dependencies to point to your fixes made it work "out of the box".
>
> ``` [...]

chrisco314, 2019-02-17:
Based on guidance from previous posters, I forked bleno-mac to restore the original sample functionality.  This is just a stopgap until these changes are merged, but it allows you to use the sample apps out of the box.  

I made the following changes:
- update bleno-mac/package.json to use @notjosh's inject-bindings branch
- copy / replicate original bleno examples directory to root level so they use the modified bleno-mac stack.

This means that at least for sake of testing, you can 

```
git clone https://github.com/chrisco314/bleno-mac.git
cd bleno-mac
npm install
cd examples/echo
node main.js
```

And then with the iOS / LightBlue app, poke at the echo server.

Webunity, 2019-02-19:
The battery service doesn't work (also doesn't work with the official repo); 
```
> cd examples/battery-service/
> node main.js 
```

```
2019-02-19 12:35:15.453 node[10406:153404] BlenoMac::Init
2019-02-19 12:35:15.453 node[10406:153404] -[BLEPeripheralManager init]
2019-02-19 12:35:15.459 node[10406:153413] CoreBluetooth BLE hardware is powered on and ready.
(node:10406) Warning: N-API is an experimental feature and could change at any time.
on -> stateChange: poweredOn
2019-02-19 12:35:15.460 node[10406:153404] BlenoMac::StartAdvertising
2019-02-19 12:35:15.461 node[10406:153404] startAdvertising:Battery serviceUUIDs:(
    180F
)
2019-02-19 12:35:15.467 node[10406:153412] peripheralManagerDidStartAdvertising: <CBPeripheralManager: 0x103000810>
on -> advertisingStart: success
2019-02-19 12:35:15.467 node[10406:153404] BlenoMac::SetServices
2019-02-19 12:35:15.467 node[10406:153404] napiArrayToCBMutableServices
2019-02-19 12:35:15.467 node[10406:153404] napiToCBMutableService
2019-02-19 12:35:15.468 node[10406:153404] napiArrayToCBMutableService: uuid:180F
2019-02-19 12:35:15.468 node[10406:153404] napiArrayToCBMutableCharacteristics
2019-02-19 12:35:15.468 node[10406:153404] napiToCBMutableCharacteristic
2019-02-19 12:35:15.468 node[10406:153404] napiToCBMutableCharacteristic: cUUID:2A19
2019-02-19 12:35:15.468 node[10406:153404] napiToCBMutableCharacteristic: value:(null)
2019-02-19 12:35:15.468 node[10406:153404] napiToCBCharacteristicProperties
2019-02-19 12:35:15.468 node[10406:153404] napiToCBAttributePermissions
2019-02-19 12:35:15.468 node[10406:153404] napiArrayToCBDescriptors
2019-02-19 12:35:15.468 node[10406:153404] napiToCBDescriptor uuid:2901 value:Battery level between 0 and 100 percent
2019-02-19 12:35:15.468 node[10406:153404] napiToCBDescriptor uuid:2904 value:'�
2019-02-19 12:35:15.468 node[10406:153404] *** Assertion failure in -[CBMutableDescriptor initWithType:value:], /BuildRoot/Library/Caches/com.apple.xbs/Sources/MobileBluetooth/MobileBluetooth-110.99.3/CoreBluetooth/CoreBluetooth/CBDescriptor.m:86
2019-02-19 12:35:15.472 node[10406:153404] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'Expecting NSData value type for Format descriptor'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff2d53d23b __exceptionPreprocess + 171
	1   libobjc.A.dylib                     0x00007fff547cfc76 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff2d542fd2 +[NSException raise:format:arguments:] + 98
	3   Foundation                          0x00007fff2f66d150 -[NSAssertionHandler handleFailureInMethod:object:file:lineNumber:description:] + 193
	4   CoreBluetooth                       0x00007fff2cfd7bab -[CBMutableDescriptor initWithType:value:] + 1178
	5   bleno_mac.node                      0x00000001032d1756 _Z18napiToCBDescriptorN4Napi6ObjectE + 268
	6   bleno_mac.node                      0x00000001032d1306 _Z24napiArrayToCBDescriptorsN4Napi5ArrayE + 117
	7   bleno_mac.node                      0x00000001032d0d81 _Z29napiToCBMutableCharacteristicN4Napi6ObjectE + 694
	8   bleno_mac.node                      0x00000001032d09af _Z35napiArrayToCBMutableCharacteristicsN4Napi5ArrayE + 117
	9   bleno_mac.node                      0x00000001032d0821 _Z22napiToCBMutableServiceN4Napi6ObjectE + 269
	10  bleno_mac.node                      0x00000001032d05f8 _Z28napiArrayToCBMutableServicesN4Napi5ArrayE + 117
	11  bleno_mac.node                      0x00000001032cd3e0 _ZN8BlenoMac11SetServicesERKN4Napi12CallbackInfoE + 338
	12  bleno_mac.node                      0x00000001032cec4b _ZZN4Napi10ObjectWrapI8BlenoMacE29InstanceMethodCallbackWrapperEP10napi_env__P20napi_callback_info__ENKUlvE_clEv + 129
	13  bleno_mac.node                      0x00000001032ceb85 _ZN4Napi10ObjectWrapI8BlenoMacE29InstanceMethodCallbackWrapperEP10napi_env__P20napi_callback_info__ + 39
	14  node                                0x0000000100a59353 _ZN12_GLOBAL__N_16v8impl23FunctionCallbackWrapper6InvokeERKN2v820FunctionCallbackInfoINS2_5ValueEEE + 283
	15  node                                0x0000000100216732 _ZN2v88internal25FunctionCallbackArguments4CallEPFvRKNS_20FunctionCallbackInfoINS_5ValueEEEE + 466
	16  node                                0x00000001002780e0 _ZN2v88internal12_GLOBAL__N_119HandleApiCallHelperILb0EEENS0_11MaybeHandleINS0_6ObjectEEEPNS0_7IsolateENS0_6HandleINS0_10HeapObjectEEESA_NS8_INS0_20FunctionTemplateInfoEEENS8_IS4_EENS0_16BuiltinArgumentsE + 896
	17  node                                0x0000000100277669 _ZN2v88internalL26Builtin_Impl_HandleApiCallENS0_16BuiltinArgumentsEPNS0_7IsolateE + 281
	18  ???                                 0x00000571b1d046fd 0x0 + 5985872660221
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6
```
Is that 'as expected'?

codeofdiego, 2019-03-04:
@chrisco314 Sorry for the late response. I was traveling. Anyway, that was for my own node project, so I added a new package file. I'm still using those branches successfully.

allnash, 2019-03-04:
@codeofdiego what's the ETA on this fix getting into release?

petrbela, 2019-04-21:
Just a note, if you're using the temporary solution

```js
"dependencies": {
    "bleno": "github:notjosh/bleno#inject-bindings",
    "bleno-mac": "github:notjosh/bleno-mac"
}
```

make sure you change all imports to `from 'bleno-mac'`. Otherwise you'll keep getting the same problem.

ankidave, 2019-04-24:
> One limitation of this approach using CoreBluetooth appears to be that bleno is unable to set the primary service UUID in the advertising data. Is this expected?

I've observed this same behavior and I believe I have a fix for it up in a pull request: https://github.com/notjosh/bleno-mac/pull/7

LukasBombach, 2019-07-04:
Thanks @petrbela gathering the stuff from here I stitched up a solution:

```js
{
  "dependencies": {
    "bleno": "github:notjosh/bleno#inject-bindings",
    "bleno-mac": "github:notjosh/bleno-mac",
    "xpc-connection": "sandeepmistry/node-xpc-connection#pull/26/head"
  },
  "resolutions": {
    "xpc-connection": "sandeepmistry/node-xpc-connection#pull/26/head"
  }
}
```

See my other post for more information here: https://github.com/noble/bleno/issues/416

madhavajay, 2019-08-07:
@LukasBombach that works perfectly! You're a legend.
I have to say after a hiatus from NPM / JS land im disappointed to see that this versioning and abandoned repo maintainer thing is still a thing. Lucky resolutions is a great fix for an annoying problem.

I wonder if there would be a way to let open source repos opt into a Code Will, where if they do not log in or reply / change the code after a period of time part control is transferred to the registry so that someone else from the community can take over and fix it. Seems like a fair thing to require of non validated open source contributors so their packages dont get stale and cause an issue for everyone in the community.

LukasBombach, 2019-08-07:
Oh I have already done that before with https://www.npmjs.com/package/react-web-component , the guy who owned the npm package before registered the name and never did anything. You can write to npm and ask for a transfer. It works quite well. 

Also I am currently working on a complete reimplementation of noble (current status: everything seems to work fine on macOS but I still need to add support for other platforms):

https://github.com/LukasBombach/sblendid/

Maybe I'll look into Bleno as well

madhavajay, 2019-08-07:
@LukasBombach amazing!! :)
I wasn't aware that NPM let you do that. It would be nice to extend that beyond the package management level to open source repos so theres less churn, but i guess a fork and a change to source repo is also pretty easy and requires less "security" issues.

Thanks again for your awesome work!

khaledalaneziq8, 2019-12-14:
Hi,

I just installed bleno and want to run it & facing this very same issue. Reading through this discussion I'm still confused about the concluding solution. It would be really nice to have a concluding answer of what steps to follow to fix. Thanks

petrbela, 2019-12-14:
FWIW, I decided to switch away from using bleno and implemented a React Native plugin with a similar API https://github.com/petrbela/react-native-peripheral
Thanks to project Catalyst, I use it to build a native Mac app for Catalina, which now works pretty much the same as my previous bleno/electron app.

LukasBombach, 2019-12-15:
@khaledalaneziq8 have you tried my solution?

https://github.com/noble/bleno/issues/404#issuecomment-508553978

Also you can use sblendid, which I made because bleno is abandoned.

https://github.com/LukasBombach/sblendid/

khaledalaneziq8, 2019-12-20:
I'm actually confused as to where to apply those lines. I added those lines in last lines in the **package.json** file (see below snippet) and ran the command **node main.js** but nothing have changed.

**"scripts": {
    "pretest": "jshint *.js lib/. test/. examples/.",
    "test": "mocha -R spec test/*.js"
  },
  "version": "0.5.0",
  "dependencies": {
    "bleno": "github:notjosh/bleno#inject-bindings",
    "bleno-mac": "github:notjosh/bleno-mac",
    "xpc-connection": "sandeepmistry/node-xpc-connection#pull/26/head"
  },
  "resolutions": {
    "xpc-connection": "sandeepmistry/node-xpc-connection#pull/26/head"
  }**

as for using sblendid, looks like it is used for BLE central, however, I'm interested in running BLE peripheral mode on my macOS.

LukasBombach, 2019-12-21:
Oh right, this is `bleno`, so `sblendid` is out. Can you post your complete `package.json` as a gist? https://gist.github.com/

khaledalaneziq8, 2019-12-22:
This is the link to the gist:
https://gist.github.com/khaledalaneziq8/fd2ec59527d6535f53fb6527e5f050f7

This is my directory structure:
├── characteristic.js
├── main.js
├── node_modules
│   ├── bleno
│   │   ├── LICENSE
│   │   ├── README.md
│   │   ├── examples
│   │   ├── **package.json**
│   │   │   ├── battery-service
│   │   │   ├── blink1
│   │   │   ├── echo
│   │   │   │   ├── characteristic.js
│   │   │   │   └── **main.js**
│   │   │   └── pizza

The package.json I posted and the main.js I try to run are highlighted in bold.

Also, find below the terminal output for running the echo program:
_**KHALEDs-MacBook-Air:echo khaledalalnezi$ cd node_modules/
KHALEDs-MacBook-Air:node_modules khaledalalnezi$ cd bleno
KHALEDs-MacBook-Air:bleno khaledalalnezi$ cd examples/
KHALEDs-MacBook-Air:examples khaledalalnezi$ cd echo/
KHALEDs-MacBook-Air:echo khaledalalnezi$ node main.js
bleno - echo
on -> stateChange: poweredOn
^C
KHALEDs-MacBook-Air:echo khaledalalnezi$**_

LukasBombach, 2019-12-22:
@khaledalaneziq8 

First of all you don't have to have to (and should never) change a `package.json` in the `node_modules` folder. You have to create your own `package.json` in your root directory. Secondly in a JSON structure you must not have duplicate keys, there must only be one key called `"dependencies"`.

Since these are issues not related to this package but to JavaScript development in general I suggest you ask this question on StackOverflow, this issue is too specific for that topic.

khaledalaneziq8, 2019-12-23:
I agree that the structure had many errors but I embarked to it after tinkering with different ways trying to make things work. Anyway, the best thing that easily worked with me out of the box was by following the below instructions. I feel it is my duty to put this summary here to help those who come later & try to install bleno. Thanks @LukasBombach for your great support & thanks to all contributors of this thread. 

> Based on guidance from previous posters, I forked bleno-mac to restore the original sample functionality. This is just a stopgap until these changes are merged, but it allows you to use the sample [...]