noble/bleno #413: npm install bleno failure

gwpantazes, 2018-09-19:
npm installation of bleno is failing on my machine.

The error can be seen [in this gist](https://gist.github.com/gwpantazes/c5ccd02e5a5fb8cf9a452e14151c6097)

Commands I have tried

- `npm install bleno`
- `npm install -g bleno`
- `npm install bleno --python=python2.7`

But there's a broken step for XpcConnection. (Again, for full log [see gist](https://gist.github.com/gwpantazes/c5ccd02e5a5fb8cf9a452e14151c6097))

```
  CXX(target) Release/obj.target/binding/src/XpcConnection.o
../src/XpcConnection.cpp:143:89: error: too few arguments to function call,
      expected 2, have 1
  ...propertyValue = object->GetRealNamedProperty(propertyName->ToString());
                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~                         ^
```

Is there a workaround for this, or does this need fixing? If you need any additional information, I'd be happy to help.

Environment:
MacOs High Sierra: 10.13.4
python --version: Python 3.7.0
npm --version: 6.4.1

jstlaurent, 2018-09-20:
That was also reported on [node-xpc-connection](https://github.com/sandeepmistry/node-xpc-connection/issues/24).

bekirdag, 2018-10-30:
I had the same issue, this worked for me: https://github.com/sandeepmistry/node-xpc-connection/issues/24#issuecomment-410697015

mark-rodgers, 2018-12-28:
> I had the same issue, this worked for me: [sandeepmistry/node-xpc-connection#24 (comment)](https://github.com/sandeepmistry/node-xpc-connection/issues/24#issuecomment-410697015)

This worked for me to install Bleno, but unfortunately Bleno has some problems on macOS Mojave (10.14.x) which is preventing my from using the library thus far :(

drewcovi, 2019-02-13:
@sandeepmistry @jacobrosenthal any chance we can update the dependency to https://github.com/jongear/xpc-connect?

This fellow forked and maintains it more routinely.